bdring/Grbl_Esp32 #2: Schematic for Grbl ESP32 Board

DirtyEngineer, 2018-07-24:
Request to release the schematic of the current Grbl ESP32 test board so others can easily build test units.

bdring, 2018-07-24:
I am on a one day adventure with the kids. I will post it in a few hours. I have a few extra BTW

buildlog, 2018-07-24:
Here is a link to the schematic.

http://www.buildlog.net/blog/wp-content/uploads/2018/07/esp32_test_schem_v1.pdf

closed 2018-07-24 by DirtyEngineer

tgirard, 2018-07-27:
I have a small 3018 engraver and a 24"X24" MPCNC and would love to help test (I'm a QA engineer here in sunny CA) any chance I could get one of those extra boards? Let me know.

DirtyEngineer, 2018-07-27:
I think I might throw together a simple ESP32 to Arduino Nano pinout converter (with additional pullups) that can be shared on Oshpark.  This way a pre-existing Grbl-Nano base board ([Example](https://www.aliexpress.com/item/I2C-IIC-CNC-Shield-V4-Engraving-Machine-3-Axis-Stepper-Motor-Driver-Expansion-Board-3-Channel/32887473387.html)) could be used for prototyping.

tgirard, 2018-07-27:
>       I can send a blank. Is that OK? I could send it, with shipping for $10.

I can do that :)
How do you want to do this?

Let me know (Yay!!!)

  

-Tim

  
  
  ------ Original Message ------
     Received: 09:27 AM MST, 07/27/2018

tgirard, 2018-07-27:
Done.
Let me know :)

buildlog, 2018-07-27:
FYI: These pullups are not on the board being sent. I tacked them on the back, then put then on the schematic to document the rework.

[image]

tgirard, 2018-07-27:
Cool.
If you could give a quick parts list I'll get ordering
Thanks
-T

buildlog, 2018-07-27:
Sent...here is the BOM. 

[esp32_test_pcb_bom.pdf](https://github.com/bdring/Grbl_Esp32/files/2236904/esp32_test_pcb_bom.pdf)

tgirard, 2018-08-02:
Just recieved the board everything and realized I got the wrong ESP 32 (I have the one with 15pins) I just ordered the correct one for the board. Got the rest of the parts from Mouser today. Going camping tomorrow through next Tuesday but after that I'll have it put together and get started. I'll post in the Wiki from here on out