bdring/Grbl_Esp32 #3: BlueTooth Serial

buildlog, 2018-07-25:
The easiest way to add this feature might be to use this library.

https://github.com/espressif/arduino-esp32/tree/master/libraries/BluetoothSerial

I am not sure how to deal with multiple communications channels. I think just echoing all the traffic on all the communication channels might be best. All messages from Grbl will be sent to both Serial and BlueTooth. Messages sent by either Bluetooth or Serial will be acted upon.

This would allow you to use either without having to change a setting or recompiling. It is up to the user to not try to stream from both at the same time. 

That way you won't get locked out of a channel, like if you set control to Serial, do you need a Serial console to change it to BlueTooth? That sounds like a problem.

buildlog, 2018-07-25:
There does not appear to be a way to use a password with that library.

buildlog, 2018-07-28:
There is also [this bug](https://github.com/espressif/arduino-esp32/issues/1537). You can mitigate the bug by sending longer strings instead of byte at a time. That would require a lot of rewriting of the reporting.

buildlog, 2018-07-29:
Things are looking good with some tests I am running. I have not run a job from a phone yet, but I am hammering pretty hard manually.

The major problem with the BlueToothSerial library is you can't use SerialBT.write(...) to write a lot of characters. You will start loosing them without knowing it after as few as 7. It is better to gather everything you want to send in a single string and use SerialBT.print(...). This meant I had to completely rewrite the report.cpp and print.cpp.  All outgoing data now goes through this code....

```
// A generic send function that everything should use, so easy to add Bluetooth, etc.
void grbl_send(char *text)
{
	Serial.print(text);
	
	if (SerialBT.hasClient())
	{
		SerialBT.print(text);
		delay(2); // possible fix for dropped characters
	}
}

```
The existing serial_read() function now also looks for BT data if SerialBT.hasClient(),

Grbl has a $I feature where you can save some build info using "$I=xxxxx". I use that to store the BlueTooth name. If it is not blank, it starts BlueTooth. That makes testing easier without rewriting settings.cpp

There is no password feature yet, but the upstream code has it now, so this library might get it soon.

buildlog, 2018-07-29:
Here is a video of where I am with the BlueTooth. 

https://www.youtube.com/watch?v=TmBBf-_LJpA

There are still some issues I would like to wrap up before I push the code. Grbl Controller is injecting some G4 P0.05 commands before things like G21 and G90 and that occasionally halts the streaming. Not sure why Grbl Controller is doing that, but the firmware should not care.

DirtyEngineer, 2018-07-29:
Excelent work!

bdring, 2018-08-04:
The issues took a while to fix, but I think it is now reasonably stable now. I think the library has some issues before it is really stable with all functions, but I have sent about 20 jobs with 100k lines of gcode over BT without issue. Also **[no password protection yet](https://github.com/bdring/Grbl_Esp32/tree/ESP32_Grbl_Bluetooth)** in the library, which really sucks, but that me change soon.

See the ESP32_Grbl_Bluetooth branch of the code.

closed 2018-08-24 by bdring