bdring/Grbl_Esp32 #4: Any vailable pins for SD card reader ?

luc-github, 2018-07-26:
Nice Job - congrats ^_^
looking at : https://github.com/bdring/Grbl_Esp32/blob/97407d1fde5b1fdf00663e0f4ed8c20f06874182/Grbl_Esp32/cpu_map.h
 it seems like there are some pins available for SPI SD card reader and may be even I2C oled screen , but I may read wrong 
Do you have the current pinout somewhere ?

DirtyEngineer, 2018-07-26:
Reference to this closed [issue](https://github.com/bdring/Grbl_Esp32/issues/2#issue)

Schematic is available [here](http://www.buildlog.net/blog/wp-content/uploads/2018/07/esp32_test_schem_v1.pdf)

luc-github, 2018-07-26:
Thank you 
wow as GPIO6 to 11 cannot be used because used by flash - few left but may be reusing some other could do the trick - 
 It may worth to try - will do later ^_^

closed 2018-07-26 by luc-github

buildlog, 2018-07-26:
Here is a snippet.  I have crossed out the pins that should probably not be used.

If you don't use some features it might be possible. Why have control buttons if you have a touch display, for example.

[image]

buildlog, 2018-07-26:
BTW: I have an unconnected SD card socket on my board to experiment with.

[image]

luc-github, 2018-07-26:
Hi thanks , GPIO0  can be used but must not be pull down at boot 
yes a serial tft could be a way 
actualy if there is a web UI no need LCD except to display the IP optionnaly (like this https://github.com/luc-github/ESP3D-WEBUI/tree/2.0 ^_^) 
The best would be to add a SD card reader so ESP32 could be a host 
Upload GCODE by WiFi and control/monitor by wifi
The issue with bluetooth I see is FW size (it become huge with BT lib) so either no space to host a webserver file or even to handle OTA but that may changed as I did test BT test few month ago