bdring/Grbl_Esp32 #7: Development Board

nhrones, 2018-07-29:
Any plans to sell the development board?

DirtyEngineer, 2018-07-29:
Referencing this [post](https://github.com/bdring/Grbl_Esp32/issues/2#issuecomment-408486191) I think Bart has a couple of the prototype development board PCB's available.

buildlog, 2018-07-29:
I have a few blanks I can sell for $10 which includes US Shipping. 

I would like to sell them to people who will actively help with the project. You can determine on your own if you qualify for that.

nhrones, 2018-07-30:
I've been busy writing a javascript Gcode server(web-app), that is served
from an ESP8266.
This app is complete, and currently running on a Chromebook as a
Chrome-App, driving a Grbl cnc router via USB => UNO
Beside file-serving this app, the ESP8266 app will use WebSocket to
transfer gcode to /from a directly coupled
(software-serial) GRBL/UNO.

This will allow any Browser to host this app and serve Gcode to the machine
over wifi via Websocket.
Browser => socket => ESP => software-serial => UNO => shield

The completed chrome-app is currently 160k (modular and un-minified).  It
has most of the capability of UGS including a canvas visualization of a
Gcode file,
This app fits nicely on the ESP8266 along with the
HTML-server/socket-server functionality, so it would be no problem for a
single ESP32-core.

I'm a retired C# / Java industrial automation developer with lots of time
on my hands.
I'm very impressed with the work you've done with the GRBL/ESP32 port.
My C skills are limited to Arduino-UNO and ESP in Arduino-IDE, but I have a
long history of successful software development experience.
I would love to work on an app-server/serial-comm core, for this project.
It would be great to eliminate the UNO from my current architecture.

I can envision the following architecture ....
   ESP Web-Server (esp32-core1) =>  browser-Gcode-sender =>
ESP-socket-server (esp32-core1) => core-to-core via semaphor  =>
GRBL(esp32-core0)
   GRBL(esp32-core0) => core-to-core via semaphor => ESP-socket-server
(esp32-core1)  => browser-Gcode-sender.

This is why I asked about the availability of a development board.
Unfortunately, at 70, my eyes are failing, and I would not be interested in
populating/troubleshooting a bare board.
II think 'll wait for a functional development board to hit the market, and
complete the 8266/UNO software in the mean-time.

Thanks, and keep up the good work.

buildlog, 2018-07-30:
@nhrones That sounds like an interesting project. When I build another for myself, I'll build a couple of them and let you know. There are only about 6 SMD parts and there are all capacitors or resistors.

You should be able to run a "virtual" machine by just running on the ESP32. Turn off homing and it should behave just fine.