bdring/Grbl_Esp32 #11: Better Serial Data Read Performance

buildlog, 2018-07-31:
Regular Grbl uses interrupts to read serial data. I did not do this on ESP32 because I expected to add multiple sources of data like BlueTooth and Wifi. I am reading via the main loop. Regular job streaming is fine, but the real time command (status, hold, etc) performance is OK, but not great.

I am experimenting with using a thread on Core0 to read the data. That appears to work better. I'll test a while longer before pushing  the code.

If you have issues related to this, let me know and I can post the push sooner.

misan, 2018-07-31:
@bdring Maybe you can push more often on a dev branch while keeping the master to a more stable, older version.

Sharing code more often may (hopefully) speed up development and testing.

buildlog, 2018-07-31:
@misan You are right. 

I made a Dev_SerialTask branch.

buildlog, 2018-08-02:
Performance of real time commands is now better.

Changes have been merged to master.

closed 2018-08-02 by buildlog