bdring/Grbl_Esp32 #12: New Grbl_ESP32 PCB

buildlog, 2018-08-02:
Here is the new PCB I am working on. I will sell it in my Tindie Store as a complete assembly, but not including the plug in modules. (Stepper drivers, ESP32, P/S and SD). Here are the features.

- It has a socket for a NodeMCU 32S ESP32 board
- It has a second set of of pads next to each ESP32 pin for easy test and hacking.
- (3) Standard stepper motor sockets with jumpers for microstep selection.
- Adjustable power supply module connected to power connector. Diode OR'd to ESP32.
-  SD module connector for SD testing (not supported by firmware yet)
- All inputs and output have individual ground connection.
- All inputs have R/C filter to reduce noise issues.
- All inputs without ESP32 pullups have external pullups.
- 3.5" x 3.5" size.

Attached are a schematic and rough placement. If you have comments, let me know, I will incorporate all reasonable requests.

[schm_esp32_cnc_test_v2.pdf](https://github.com/bdring/Grbl_Esp32/files/2254297/schm_esp32_cnc_test_v2.pdf)
[image]

DirtyEngineer, 2018-08-02:
Glad to see a development board moving forward. Is the end goal of the SD card to be able to run code from it?

On a semi related note I put together a ESP32 "DOIT" to Arduino Nano converter that will be headed to Fab tomorrow.  It adapts both the 30 and 36 pin "DOIT" boards for use with the Grbl Nano CNC boards.

[image]
[image]

DirtyEngineer, 2018-08-02:
Adding the filtering on the Inputs was a wise move. I'll be sure to do that on the next iteration of the converter board.  Will the SD version be a different branch of the code as boards based on the V1 schematic will be incompatible?

bdring, 2018-08-02:
I hope to use the same code via config.h and cpu_map.h options. The V1 does not actually wire the SD card. It just makes it easy to jumper it in.

unixbigot, 2018-08-03:
I just sent a two-axis-plus-servo board off for manufacture.   

[image]

bdring, 2018-08-03:
Nice, do you plan to run Grbl on it?

bdring, 2018-08-03:
@unixbigot I see you are using the ESP32-WROOM module. Will you have USB connectivity?

nhrones, 2018-08-03:
This board looks great. Thanks for all your hard work. I'll be sure to purchase.

unixbigot, 2018-08-03:
@bdring My board is targeted at grbl, yes.   The header below the wroom is for a usb module, the micro-usb at right is for power only.    My goal is to implement a wifi bridge compatible with Chilipeppr and laserweb.

bdring, 2018-08-03:
@unixbigot  What USB module. Is there a standard that people like to use when using a ESP32-WROOM module

unixbigot, 2018-08-03:
@bdring the defacto standard for TTP serial headers is the pinout from the [FTDI FT-232 cable](http://www.ftdichip.com/Support/Documents/DataSheets/Cables/DS_TTL-232R_CABLES.pdf).  This is the pinout used by the Arduino Lilypad, pro mini, etc.   The Wemos [CH340 Serial module](https://www.aliexpress.com/store/product/WEMOS-CH340G-Breakout-5V-3-3V-USB-to-serial-module/1331105_32664922086.html?spm=2114.12010608.0.0.5c2d3fc1v1GJg0) has a compatible pinout; my board has a header to match the Wemos pinout.

bdring, 2018-08-03:
@unixbigot Thanks

I am planning on a project using that module and I did not want to re-invent any standards.

BTW: My board is on order and should arrive in about a week.

bdring, 2018-08-08:
Got  version 2 working. 

- Cleaned up the features
- Added R/C filters to all inputs
- All major circuits are now low cost plug in modules.
- Added pullups to the control switch inputs. (ESP32 does not have internals for these)

It will be on Tindie next week. Price TBD
I'll publish the source files too.

[image]

bdring, 2018-08-21:
The PCB is now fully tested and for sale on Tindie.

https://www.tindie.com/products/33366583/grblesp32-cnc-development-board/

[image]

DirtyEngineer, 2018-08-21:
Bart that board looks fantastic!.  I've got something to show off as well.
[image]
[image]
[image]
[image]

derFrickler, 2018-10-03:
Nice work! 
Maybe the ESPduino-32 together with the CNC-Shield could although be an nice and easy option. 
https://github.com/SmartArduino/SZDOITWiKi/wiki/ESP8266---ESPduino-32

The GRBL CNC Shield for the Arduino Uno is dirt cheap and available everywhere. 
https://www.google.com/search?q=grbl+shield&client=ubuntu&channel=fs&source=lnms&tbm=isch&sa=X&ved=0ahUKEwjFleiG1undAhUJ_aQKHW8wCT0Q_AUICigB&biw=1575&bih=974

xsiriusx, 2018-11-08:
> Here is the new PCB I am working on. I will sell it in my Tindie Store as a complete assembly, but not including the plug in modules. (Stepper drivers, ESP32, P/S and SD). Here are the features.
>
> * [...]

easyeda link?

AlmightyFrog, 2018-11-19:
Currently I am trying to get it running on a ESP-32S board free wired to an Arduino CNC Shield.
(Sadly currently now working as I don't get it to compile both in Arduino IDE and platform.io)

@derFrickler I didn't know about ESPduino 32 but that would be really a good solution for easy access.
But what about the missing pullups on most inputs which seem to be mandatory (if I understood documentation correctly)?

bdring, 2018-11-19:
@AlmightyFrog  Only GPIOs 34-39 don't have pullups. It is easy to add them externally.

Harvie, 2018-11-24:
What about start/hold/stop buttons? Are there any pins left for that?

bdring, 2018-11-24:
@harvie

What PCB?

On my PCB there are pins for those.

bdring, 2018-12-10:
@DeanRM I'll take that into consideration on the next version. Do you have a link to a datasheet on the hall sensors.

DeanRM, 2018-12-10:
https://www.mpja.com/download/a3144eul.pdf.
Any open collector hall sensor will work.  Three different types hall sensors - N pole, S pole or both.  I use two in parallel on each axis to detect the moving magnet between them.  Once detected - Grbl can back off 1mm for homing which is more accurate than what is needed with a switch.

bdring, 2018-12-10:
It would be safer to put 3.3V. Is that OK?

DeanRM, 2018-12-24:
5v would be better.  I am worried about voltage drop on the long wires.  Smaller gauge wires (no control over what people use) - higher resistance - less voltage available on both the supply and signal lines.  If 3.3v is necessary, then a warning to use optocouplers may be needed or you could add them into the design to isolate input signals from whatever is used.  There are 3.3v hall sensors available, but voltage drop on wires may put them into a gray area.

Harvie, 2019-01-03:
Arduino nano socket is much smaller and should fit inside of nodemcu esp32 socket. So we can have both.

bdring, 2019-01-03:
I have a new (Rev 2) 2D pen/laser machine controller on order. The most significant changes are a 5V PWM for laser power control and a dual width socket for the ESP32 module.  

[image]

jschoch, 2019-01-11:
what is the max voltage input for board v2.1?

GandalfOp, 2019-02-07:
I'm trying to build a "SelfRunning" machine using your pcb, I could upload the gcode to the sd, run it from a Gcode sender, and I would like to do that without a computer, can I add any code to make the "start" Button automatically send the Gcode to my steppers?

Thanks!

bdring, 2019-02-07:
Not with the current firmware. 

I have wanted a similar feature for your type of operation and other macros like home.

I think one way would be to create another comm channel, like serial, bluetooth and wifi use.

As a quick hack you could just push data on to the serial buffer.

GandalfOp, 2019-02-07:
That would only allow us to print one gcode (the one named autostart.gcode) but it would make the job in my opinion.
Having a "plug and play" cnc machine would be great to be honest! My goal is to print always the same gcode, and be able to place it wherever I want (I'm talking about a small machine with nema8, not something heavy ^^)

AbySet, 2019-06-14:
I was looking at the first post pdf file (schematic) and because my electronic knowlege is average, i still confused about some pins are shared for memory card reader and machine controle outputs:
Is the memory card reading during the machine work ?
If so, how can CS of card reader still enabled if i decide to reverse the spindle direction.
If not, how (meaning when) the ESP32 reads gcode file, during the machine work or before when assuming its puting it in internal memory ?
I'm planing to begin a project with 4 steper motors and maximum monitoring system thats why i'm asking about memory card reader part.
Thank you.

bdring, 2019-06-14:
[That controller](https://www.tindie.com/products/33366583/grbl_esp32-cnc-development-board-v35/) allows you to choose what functions you use. You cannot use more than one per pin.

The SD card uses SPI, so you can put multiple SPI peripherals on that. I did that with the TMC2130 drivers.

bdring, 2019-06-14:
This is that [TMC2130 controller](https://github.com/bdring/Grbl_ESP32_TMC2130_Plotter_Controller)

dpharris, 2019-06-14:
Nice!  You do great work.  I watched some of your videos.

David

AbySet, 2019-06-14:
Of course, its clear to me now after i took a look at the links.
Thank you.

closed 2019-11-05 by bdring

aecampo, 2021-06-26:
@DirtyEngineer I am looking for the ESP32 to Nano adapter board files to create my own adapter, do you have any boards made in eagle?

aecampo, 2021-06-26:
> Glad to see a development board moving forward. Is the end goal of the SD card to be able to run code from it?
>
> On a semi related note I put together a ESP32 "DOIT" to Arduino Nano converter that [...]

I am looking for the ESP32 to Nano adapter board files to create my own adapter, do you have any boards made in eagle?

guestisp, 2022-04-14:
Something available with full sources (i need some customization) with almost the same form factor as cnc shield v4 but for esp32 ?

I need mostly a cnc shield v4, with some minor changes