bdring/Grbl_Esp32 #15: Leaving door pin floating causes crash loop

unixbigot, 2018-08-04:
When I first compiled and deployed to a breadboard, I received a crash loop.

After fitting pullups to button pins things settled down.

I'll update the README.

unixbigot, 2018-08-04:
Possibly due to bugs in early revs of the ESP32 silicon.

bdring, 2018-08-04:
I think all ESP32 silicone is that way.  Here is a snippet from the datasheet.

[image]

bdring, 2018-08-04:
It might be a good idea to add an optional way to turn off the use of control pins. Many (most?) people don't use them and this would be a fast way to fix this issue in firmware.

@unixbigot What do you think about this? Maybe a **#define USE_CONTROL_PINS** in **cpu_map.h**.

bdring, 2018-08-21:
@unixbigot 

I took another look at this and decided to make this an option.

if **#define IGNORE_CONTROL_PINS** is not commented out (the default) the switches will be ignored.

```
uint8_t system_control_get_state()
{
	#ifdef IGNORE_CONTROL_PINS
		return 0;
	#endif
...
```

I'll include this in a future push.

bdring, 2018-08-24:
Closing latest push adds plenty of options for door and control pins.

closed 2018-08-24 by bdring