bdring/Grbl_Esp32 #17: (USG) DRO updating is sluggish

tgirard, 2018-08-06:
I'm reporting this thinking that maybe it's a parameter with the new board that's not quite right.
This is compared to a .9 Grbl Board
The DRO's get significantly behind when using the new board with the same Reporting Setting (3)
Also, USG takes longer to come out of locked mode when the board has completed a move. With the .9Grbl board, control is handed back to the user fairly quickly. Here, The delay is very noticable and becomes annoying quickly 

Esp32 CNC board v1. > 24v
8825 Steppers at 400mv vRef
No Homing, No Limits

bdring, 2018-08-06:
Are you using the latest master branch of the code?

I noticed sluggish performance in responding to the "?" command due to the fact that the reading the serial is not interrupt driven. I changed that in the most recent version. [See this closed issue](https://github.com/bdring/Grbl_Esp32/issues/8).

I'll give UGS a try and see if I get the same issues.

bdring, 2018-08-07:
I have compared UGS job durations to other senders on a few short jobs and they appear to take the same time. The motion performance does not appear to be affected. When the DRO's are sluggish, the motion is still running smoothly.

I used the Bluetooth to capture the data. UGS used a Bluetooth serial port and I had an Arduino IDE open on the USB serial port. All Bluetooth data is echo'e to the serial port before any processing. 

Here is a portion of the data while the DRO's are sluggish. At the beginning things are fine, then there is a big gap of ?'s, then ?'s return. If UGS runs fine on other Grbl, something could be different and confusing UGS.

?<Run|WPos:0.709,0.312,3.810|Bf:0,38|FS:703.150,0.000|Pn:PRHS>
?<Run|WPos:3.399,1.531,3.810|Bf:0,38|FS:1360.787,0.000|Pn:PRHS>
?<Run|WPos:8.276,3.719,3.810|Bf:0,38|FS:1457.710,0.000|Pn:PRHS>
?<Run|WPos:12.296,5.521,3.810|Bf:0,38|FS:767.191,0.000|Pn:PRHS>
ok
?<Run|WPos:14.089,6.323,3.810|Bf:0,60|FS:78.394,0.000|Pn:PRHS>
G1X12.306Y9.784F762.0
?<Run|WPos:14.207,6.375,3.180|Bf:0,38|FS:304.800,0.000|Pn:PRHS>
?<Run|WPos:14.207,6.375,2.140|Bf:0,38|FS:304.800,0.000|Pn:PRHS>
?<Run|WPos:14.207,6.375,0.980|Bf:0,38|FS:304.800,0.000|Pn:PRHS>
?<Run|WPos:14.207,6.375,0.060|Bf:0,38|FS:304.800,0.000|Pn:PRHS>
ok
G1X8.927Y6.897F762.0
?<Run|WPos:14.207,6.375,-0.970|Bf:0,39|FS:181.081,0.000|Pn:PRHS>
ok
G1X8.224Y6.214F762.0
ok
?G1X8.399Y6.020F762.0
ok
G1X8.923Y5.883F762.0
ok
G1X14.205Y6.374F762.0
G1Z3.810F304.8
ok
G0X20.083Y42.239
ok
G1Z-1.000F304.8
ok
G1X21.258Y44.079F762.0
ok
G1X22.114Y45.096F762.0
ok
G1X23.123Y46.049F762.0
ok
ok
G1X25.771Y48.109F762.0
ok
G1X22.282Y51.598F762.0
ok
G1X18.792Y54.929F762.0
ok
G1X18.713Y56.555F762.0
ok
G1X18.203Y58.872F762.0
ok
G1X17.855Y59.814F762.0
ok
G1X16.969Y61.229F762.0
ok
G1X15.791Y62.412F762.0
ok
G1X14.389Y63.302F762.0
ok
G1X13.625Y63.620F762.0
ok
G1X12.829Y63.842F762.0
ok
G1X10.918Y64.032F762.0
ok
G1X9.021Y63.922F762.0
ok
G1X8.592Y63.807F762.0
?<Run|WPos:2.719,15.281,-1.000|Bf:0,38|FS:762.000,0.000|Pn:PRHS>
ok
G1X8.351Y63.648F762.0
?<Run|WPos:2.739,13.031,-1.000|Bf:0,39|FS:198.733,0.000|Pn:PRHS>
ok
G1X8.300Y63.451F762.0
?<Run|WPos:3.251,13.031,-1.000|Bf:0,40|FS:626.718,0.000|Pn:PRHS>
ok
G1X8.443Y63.224F762.0
?<Run|WPos:5.123,14.552,-1.000|Bf:0,40|FS:762.000,0.000|Pn:PRHS>

bdring, 2018-08-07:
Oops, maybe there is a missed one. Near the top of the gap is...?G1X8.399Y6.020F762.0. That ? has no response.

Most gcode senders send ? on a specified frequency regardless of a response, maybe UGS waits for a response and takes a while to try again.

Maybe there is a race condition and the flag to send the report is cleared early.

tgirard, 2018-08-10:
Hey Bart,
that change made all the difference in the DRO performance. I did a whole bunch of back and forth on all three axis's and UGS performed the same as the .9 board (No-Load)
Closed as Fixed

closed 2018-08-10 by tgirard