bdring/Grbl_Esp32 #18: UGS Setting Idle time to 255 (Lock steppers)  8825's get hot fast at low current (400mv Vref)

tgirard, 2018-08-07:
I don't have a comparison but the 8825's are very hot to the touch when the steppers are locked with no load
$1 = 255    (Step idle delay, milliseconds)

I have the Vref under 1 amp (24v main power, 400mv at the Pot on the 8825 so a current of .8 Amps).
my laser thermometer is showing the drivers are reaching 100 degrees F.

I'm not 100% that everything is super accurate but they are definitely hot (I have the heat sinks on)

I am just concerned if they are this hot at .8, where will they be when drawing 1.5 or 2 amps holding a load.

Again, I'm not sure if any of this could be caused by the board

bdring, 2018-08-07:
They will get hot. If you hear a ticking sound from the motors, they could be shutting down.

100°F is not too hot. It is that hot outside here :-). They should be able to run above 125F. 

It has nothing to do with the board.  You should change they value below 255 if you want high current. Motors draw the most power when idle.

tgirard, 2018-08-08:
Yeah, I haven't run the current up and tried holding the motors. I'll try that and see if they start to really cook. FYI, It's not the motor that's heating up but the driver

bdring, 2018-08-08:
@tgirard Yes, I was talking about the drivers too, but if the drivers overheat they will cycle on and off. That will cause a ticking sound in the motors as the coils pop on and off.

Motors can get very hot without damage. Most of the time hot motors will damage something else like plastic parts, pulleys and belts. 

I melted a motor through a plastic table once during testing. Motors was fine, table not so much.