bdring/Grbl_Esp32 #22: WiFi Feature Discussion

bdring, 2018-08-21:
It is time to start thinking about how to implement WiFi. I see two basic areas to start talking talking about.

**WiFi Network Type**
- It is an access point (like some action cams)
  - Pros
    - No setup
    - Does not need an existing network
  - Cons
    - You have to switch away from your existing network to use it.
- It is on an existing network
  - Pros
    - You don't have to switch between networks often
  - Cons
    - You need an easy way to give it the SSID and PWD
    - You need an existing network

**How do you use it**

- [Web Socket](https://en.wikipedia.org/wiki/WebSocket)  like [LaserGRbl Supports](https://github.com/arkypita/ESP8266-SerialWebSocket)
  - Pros
    - Some software already supports it.
  - Cons
    - This is streaming and the connection needs to be robust.
- Web Interface like very basic [Octoprint](https://octoprint.org/)
  - Pros
    - Totally self contained.
    - No apps required
    - No worry about cross platform issues. You just need a browser.
  - Cons
    - Complex project
    - Could use a lot of memory and performance

luc-github, 2018-08-21:
I could port https://github.com/luc-github/ESP3D-WEBUI for GRBL-ESP32
It is originaly for ESP8266 / ESP32 addon board (like on http://www.panucatt.com/azteeg_X5_mini_reprap_3d_printer_controller_p/ax5mini.htm)  but I am already porting it for Marlin-ESP32 FW : https://github.com/luc-github/Marlin/issues/1

 I think porting for GRBL-ESP32 will be easier as it use ESP32 SD lib unlike Marlin-ESP32

Not sure if there is space also for Serial commands to configure WiFi, on Marlin I used the reprap ones: https://raw.githubusercontent.com/luc-github/Marlin/esp32/Marlin/src/HAL/HAL_ESP32/ESP32_GCODE.txt

Could be a good leverage as lot work is already done.

One comment using AP mode - it is good for initial configuration but it can be easily flooded as computers do a lot of queries in background - many application do some auto check and can make AP busy for nothing as it is the gateway or need to set a fake gateway IP to bypass them

bdring, 2018-08-21:
@luc-github That is very impressive. It seems like a good starting point. Thanks for the tip on AP mode.

luc-github, 2018-08-22:
I will check Grbl_Esp32 code ASAP but in theory need 4/5 entries points if I refer to my experience  with marlin porting : 
1 - an entry point to init wifi part
2 - an optional entry point in main loop for not critical actions, not mandatory just handy
3 - an optional entry point in gcode interpreter for a set of commands if you want to manage wifi by serial, not mandatory but really handy
4 - a function to inject GCODE manually - may be this function exists already, sorry I did not checked yet
5 - an entry point to catch what is sent to Serial - allowing to get result of commands injected by previous function

I call entry points because  if they are defined then any lib, not only ESP3D, could be hooked on them giving flexibility - it will be also an easy way to disable wifi for debug reason for example 

just my 2 cents

bdring, 2018-08-22:
Great. I would suggest using the SD card branch. Having Bluetooth, Serial and SD in the same code might make the existing entry points more obvious to you. 

1.  Some of the basic inits() are in the setup() function. There are also some in the beginning of the loop() function. The beginning of the loop() only runs after a Grbl reset. At the end of loop(), protocol_main_loop();  runs continuously until a reset. 

If you only want to init() once, put it in setup. If you want to restart something when Grbl does a soft reset, put it in the top of loop()

2. protocol_main_loop() is a good place for non critical actions. This is where Grbl reads from the Bluetooth and Serial ports.

3. gc_execute_line(char *line) is a good place to inject gcode. It is also a good place to put hooks into the wifi code.

4. gc_execute_line(char *line) might do what you want.

5. Incoming serial data read by a task,  serialCheckTask(void *pvParameters).  basic "realtime" commands are checked and then the data is put in a buffer. The buffer is then read by protocol_main_loop().

luc-github, 2018-08-22:
Excellent ^_^ - thank you - looks like everything is ready to start :smile_cat:

luc-github, 2018-08-22:
Hi I have started reading code and I have question:
about 5 - actually is Serial output which is needed, not input but it is ok because  I found : `void grbl_send(char *text)` which looks like what I need  

but I see also several part using directly `Serial.printf` or `Serial.print` - any reason that print.cpp mainly does not use grbl_send ?

bdring, 2018-08-22:
You are correct. grbl_send(0 should be used. I will fix that soon.

Right now all data, regardless of source is treated the same way. I think eventually grbl will need to know where the data came from. 

Like ... If Bluetooth asks for status, only return status to Bluetooth.

luc-github, 2018-08-22:
yes knowing from where the command come from would be nice, so answer would go to same 'pipe' and not poluate other 'pipes' (Serial / BT / WEB/ etc...) but this may complexify the code - I did it in ESP3D https://github.com/luc-github/ESP3D/blob/2.0/src/espcom.h but ESP3D has a lot of 'PIPES'
```
typedef enum {
    NO_PIPE = 0,
    SERIAL_PIPE = 2,
    SERIAL1_PIPE = 3,
    SERIAL2_PIPE = 4,
#ifdef TCP_IP_DATA_FEATURE
    TCP_PIPE = 5,
#endif
#ifdef WS_DATA_FEATURE
    WS_PIPE = 6,
#endif
#ifdef ESP_OLED_FEATURE
    OLED_PIPE = 7,
#endif
    WEB_PIPE = 8,
    PRINTER_PIPE = 9
} tpipe;

```

bdring, 2018-08-22:
That might be why I didn't get the [FILE:...] response in Grbl Controller (Bluetooth) 😄

cemir, 2018-09-13:
Hi , would like to help here too, 

I'm currently in the process of creating a drawing machine (kind of AxiDraw clone) with an integrated user interface.

So my id was to build some kind of webapp running on the ESP, allowing a user to upload GCODE to SD card and providing a way to start the print from sd. + simple controls (like a very simplified octoprint)

The only "negative" point : I have no idea on how this will affect performance, so I'll try to create a POC first.

Also, on boot, the board will try to connect to a configured SSID, if failed or no SSID configured, it will boot in AP mode, allowing users to connect to it and configure it. 

Let me know if I can help with the wifi/webserver part please let invite me to slack.

luc-github, 2018-09-13:
Hi the WiFi part is currently here : https://github.com/luc-github/Grbl_Esp32/tree/SyncWebServer
still need so tuning but almost feature are there

the Web UI is here : https://github.com/luc-github/ESP3D-WEBUI/tree/GRBL_ESP32, I am now working on it and hope to finish next week

please feel free to test and feedback

bdring, 2018-09-13:
@cemir   [Here is an invite link.](https://join.slack.com/t/buildlog/shared_invite/enQtNDI2NDI4NDkxNTg4LWEzMTdlODU4ZDRmMjQ2ODU2NWM4MGFiZDZjM2NmN2YxYjgxZDQxOGM3YmEzNDNlOTgxNTZmMDVjOTAyMjg5ZmY)

The WEBUI is getting very close to completion. It did take a lot of experimentation to make sure nothing affected the performance.

jeffeb3, 2018-09-15:
Have you seen cnc.js? I don't know if it could be compressed small enough to live on the esp, but it would be also be awesome if you made a cnc.js plugin (or maybe it already supports the websockets? Maybe use that for PC/Linux application? I would think loading and running gcode from the esp/sd card would be more reliable than streaming over websockets during an operation.

bdring, 2018-09-15:
@jeffeb3 Thanks for that suggestion. 

We can probably support cnc.js, but not put it on the ESP. It is probably better to allow programs like that to connect using their existing websocket method.

luc-github, 2018-09-15:
about cncjs 
the src + static directories compressed give  1 436KB which is really over current SPIFFS capacity, so need some work to adjust to fit size I think

I also think it need also a big rewrite to be executed from ESP itself and monitor SD printing instead of being  a host because I fully agree that : 
> I would think loading and running gcode from the esp/sd card would be more reliable than streaming over websockets during an operation.

bdring, 2018-09-16:
Closing, starting as a fresh issue about ESP3D Web-UI

closed 2018-09-16 by bdring