bdring/Grbl_Esp32 #23: Issue with eeprom library

MCuserme, 2018-08-23:
I am having a lot of trouble compiling this firmware.
I get a fatal error: EEPROM.h: No such file or directory  for a library import line in Grbl.h.

I was thinking maybe library is missing from arduino core libraries, but it was there.

I also copied that lib in the arduino projects folder and in this project folder to be safe, but still not compiling.

I then commented that import line in grbl.h and placed it in Grbl_Esp32.ino, now it passed import but since library is imported before grbl.h I get error that EEPROM.xxx is not defined SO I moved it after import of grbl.h and the "fatal error: EEPROM.h: No such file or directory" error returned.

**My guess is that EEPROM.h library is conflicting with eeprom.h that is in this project.**

I am probably making some stupid mistake so help would be AWESOME 💃

bdring, 2018-08-23:
Are you able to run any of the EEPROM examples for ESP32?

[image]

MCuserme, 2018-08-23:
Yes, I just compiled one and it was successful.

bdring, 2018-08-23:
I don't know.  Nobody else has had this issue.

Your guess about eeprom.h might be worth trying.  Try renaming the Grbl_ESP32 eeprom.h and eeprom.cpp to grbl_eeprom.h and grbl_epprom.cpp and change the #include