bdring/Grbl_Esp32 #24: Control of a VFD Spindle over RS485, ie a second serial port.

davehines91, 2018-09-01:
This is a port from my unpublished, but well used work using grbl-Mega

Background : 
I currently use grbl-Mega to control a DIY CNC router  with a work area of about 600mm x 900mm.
The spindle I currently use is an 800w spindle controlled by a 1.5KW Huanyang Variable Frequency Drive (HY01D523B) . I was looking for a more convenient platform than the Mega , and stumbled upon the ESP32 port.

Testing : 
So far I've tested this with the ESP32 dev module in a breadboard , controlling a single axis , and a [RS485 module](https://www.ebay.co.uk/itm/RS485-Module-MAX485-TTL-RS-485-for-Arduino-Raspberry-PI-DIY-UK-Stock-/111723044954) , connected via a [levelshifter](https://www.adafruit.com/product/757) breakout board to the ESP32 . (Note , the 3.3v version did not work well for me ) .
This setup was used to connect to the VFD over the RS485 interface.

Implementation: 

The RS485 link is a two wire differential half duplex interface. The controller needs to assert a send signal 
as it transmits , I've controlled this with a timer interrupt ( I've used timer 3 ) . The timer is restarted as each character is transmitted , so the transmit is held high for 4mS after the last character is send , the response is then awaited for by a task , shamelessly cloned from the main serial task. This task checks for the correct response from the VFD , and in the case of the speed request , holds up any further command processing , until the spindle has reached the required speed. If it fails to reach the required speed, the motor is sent the stop command, and an alarm ( 10) is raised.

All the interface to grbl has been implemented  within  spindle_control.cpp and is enabled or disabled via the Macro RS485_HUANYANG_MOTORCONTROL  , currently defined in config.h.  The pins used are defined in cpu_map.h , these could be the same three pins currently used by grbl for pwm control, but I've left this code in place in parallel at the moment.

Repo currently at https://github.com/davehines91/Grbl_Esp32

davehines91, 2018-09-01:
I'll try and draw up one tomorrow.

davehines91, 2018-09-02:
[ESP32-VFD.pdf](https://github.com/bdring/Grbl_Esp32/files/2343313/ESP32-VFD.pdf)

Here is an approximate schematic of my setup.   I've used the adafruit level shifter , but I couldn't find an eagle library for it , so I've drawn it up with the [sparkfun](https://www.sparkfun.com/products/12009) one. The pinout is slightly different .

davehines91, 2018-09-02:
:) 
I couldn't find any transceivers with auto direction control at the time I first did this.
It was also a exercise in interrupts on the arduino for me.  and likewise on the ESP32.
It was also more configurable if different baud rates were to be used. 
More instructions currently being typed up.

davehines91, 2018-09-02:
To run the motor use the G-code command : 
  M3 S10000
To Stop the Motor use:
  M5 S0
grbl doesn't seem to like commands  like M5 or M3 without a speed .

Various debugging messages, written to Serial1 , can be enabled by defining DEBUGSERIAL in VFD.h

I've hard coded the BAUD rate to 38400 binary 8N1 , PD164 needs setting to 3 for this
baud rate. similarly PD165 should be set to 3 for binary 8N1 comms.

PD01 and PD02 need setting to 2 to respond to stop/start and speed commands
over RS485.

I think this is the last of the things I set up , but most was done along while ago , and while my memory is quite good .....

bdring, 2018-09-02:
I think RPM is modal, like feedrate.

S500 ;set rpm but don't turn on spindle
M3 ;turn on CW
S1000 ; change rpm
M5 ; turn off
M3 ; turn on at 1000

You can see all modal values at any time by sending $G

davehines91, 2018-09-02:
I've ordered a couple on the 485 converters you list above off ebay from china, for about $2 each .
I'll try them when they arrive, The ESP32 is little short on available pins , so any we can save will be a bonus

On the subject of saving pins , do you know if it is possible to allocate inputs and outputs , to unexposed GPIO pins  ,( 20,24,28,29,30,31) (37,38 have no pull up resistor , so I wouldn't try these ) so their pins could be used for other purposes  , I'm thinking 4th  axis  as an example.

misan, 2018-09-02:
Only TX is really needed, as the RS-485 driver may be kept enabled all the
time and RX can be ignored. Of course, if you do that there is no feedback
from the VFD but it is definitely a possibility to save pins.

davehines91, 2018-09-02:
I considered that when I first implement the motor control on the Mega, but I consider it safer to wait til the motor has run up to speed before continuing. so feedback is required IMHO.

misan, 2018-09-02:
@davehines91 let me elaborate a bit more: for starting/stopping or setting the RPMs only TX is needed if you agree on giving up feedback. Otherwise, you need 3 I/O pins and can raise an error condition when, for example, the spindle is not starting up as requested so a tool will not be broken. 

I guess the module above does use TX signal activity to trigger a timer to enable the driver output thus saving one output pin.

davehines91, 2018-09-02:
@misan I agree, 
I'm describing what I have implemented for grbl-Mega and used successfully for about 18 months, and have just ported to Grbl_Esp32.
Bart has asked me to share my work , which I'm more than happy to do.

bdring, 2018-09-02:
@davehines91  unexposed GPIO pins.

I don't know. It is probably easier to use something like `#ifdef PROBE_PIN` to free up some pins.

Ultimately the use of an I/O expander will be supported for low speed I/O, but the initial goal is to get all features working for a basic machine in a single chip (ESP32) solution.

davehines91, 2018-09-02:
@bdring  #ifdef, yes , I would agree , I was just trying to make minimal changes.

I like the idea of an IO expander , though I'm more than happy with your port so far .
I look forward to actually cutting some plastic with it when your board arrives.

misan, 2018-09-02:
@davehines91  I was not so careful. I started by checking VFD response http://fightpc.blogspot.com/2014/10/vfd-control-with-arduino-using-rs485.html but ended up ignoring it in a patch I made for Marlin firmware. Someone suggested me to forward the patch for GRBL which I did but that was considered too specific a feature.

davehines91, 2018-09-02:
@bdring 
> I think RPM is modal, like feedrate.
I've just pushed a change to my repo which fixes this behaviour 
Thanks for pushing me  :)

Harvie, 2020-06-24:
Can you please send your huanyang VFD RS485 code to this issue? https://github.com/terjeio/grblHAL/issues/68