bdring/Grbl_Esp32 #26: Nano form factor adapter board

ggallant571, 2018-09-03:
A couple of weeks ago there was a post about an adapter board that would allow an ESP32 module to be plugged into an Arduino Nano form factor. I am currently using STM32 boards in the Nano package and wondered if the board design is available?

I also have a laser etcher that is built in a Core-XY configuration. Works great with a standard Arduino and cheap 4 motor shield. Only issue is the speeds appear to be off by the square root of 2. Easy enough to compensate but would like to migrate this maching to the ESP and wondered if anyone has solved this problem.

bdring, 2018-09-03:
[This is the issue](https://github.com/bdring/Grbl_Esp32/issues/12) where that board was mentioned.

If the speed if off, it makes sense that is the amount. The work is basically rotated 45% for core XY. I have not tried any core XY on this code yet, but I would guess it is the same as Grbl.

DirtyEngineer, 2018-09-04:
@ggallant571 Are looking for an ESP32 to Nano adapter board or the board files to create your own adapter?  I have plenty of extra converter boards if you want one. The board is made in Eagle.

ggallant571, 2018-09-07:
I am interested in getting some of these boards. Prefer to purchase them from the designer but will order if necessary. Is the schematic and art work available?

On another subject - does replying to an email get posted to this forum?

DirtyEngineer, 2018-09-07:
Attached is a pdf of the current schematic.

[ESP32 to Nano V1.0.pdf](https://github.com/bdring/Grbl_Esp32/files/2362132/ESP32.to.Nano.V1.0.pdf)

I can send a couple bare pcb boards your way for the cost of postage.

ggallant571, 2018-09-07:
Please send to:

   George Gallant
   39 Great Oaks Lane
   Otisfield, ME 04270  USA

In include or post return address and I will send cash.

ggallant571, 2018-09-07:
I would also be interested in purchasing a couple MIC5219 chips.

carl1961, 2019-06-21:
@DirtyEngineer I would like to purchase your Esp32 to nano board , I want to see if it will work on the ardunio nano io shield .

closed 2019-11-05 by bdring

Jortega00, 2021-02-19:
where can I buy this adapter? I'm looking for one of this for a school project

aecampo, 2021-06-26:
@DirtyEngineer  I am looking for the ESP32 to Nano adapter board files to create my own adapter, do you have any boards made in eagle?