bdring/Grbl_Esp32 #27: Is there a spindle direction pin assigned?

MCuserme, 2018-09-09:
I was looking at cpu_map file to get pins I have to use and I could not find spindle direction pin.
On atmega version it is usually pin 13.
So is it possible to assign spindle direction to a pin or not yet?

Thank you!

bdring, 2018-09-09:
Currently the focus is getting Bluetooth, the SD card and Wifi working using only I/O on the CPU. Everything is done except wifi (almost done).

The spindle direction pin on the [reference hardware design](http://www.buildlog.net/blog/2018/08/grbl_esp32-cnc-development-board/) is being used for the SD card, so getting spindle direction working is not a priority. 

You could probably use GPIO16 (currently flood coolant), or GPIO21 andd 22 (probably an LCD in the future)

BTW: What application needs spindle direction?

MCuserme, 2018-09-09:
I have a VFD for a spindle with which I control speed and direction. It is not critical to have a spindle dir controlled by software but it is nice to have.

I am currently in process of building a cnc machine where I use 2.2kw spindle.

I do not plan to use SD card or Wi-Fi or anything of that sorts so what code should I add to assign spindle direction to a specified pin.

Thanks

MCuserme, 2018-09-09:
BTW if you dont have time to deal with this, you don't have to I can live without it.

bdring, 2018-09-09:
I'll give it a look soon.  Do you use the flood coolant control? That would be the easiest to disable.

Just curious, what machining operation do you do with the spindle in reverse.

MCuserme, 2018-09-10:
It is true that is is very rare you need CCW spinning but there are endmills and drills that need to spin CCW.
I dont use flood coolant control.

bdring, 2018-09-10:
OK, I am adding spindle direction and enable control. Both are optional and can be assigned to any available I/O pin.

As part of this feature I am changing the way coolant controls work. To enable either (mist or flood) coolant control you simply assign it a pin.  

If this method works well, I may use it to enable more axes.

It should not take too long to add and test.

bdring, 2018-09-10:
Done. Try is on the dev branch.

MCuserme, 2018-09-10:
Thanks, I will try it tomorrow.

closed 2019-10-14 by bdring