bdring/Grbl_Esp32 #30: ESP32 Web-UI

bdring, 2018-09-16:
This issue is to discuss the Wifi and ESP3D Web-UI features being added. This topic is for **very high level discussion onl**y. Daily discussions are being done on the Slack channel (ask for an invite)

[Here is a wifi page to show off sme of the features](https://github.com/bdring/Grbl_Esp32/wiki/ESP3D-WEBUI-for-Grbl_ESP32)

[image]

bdring, 2018-09-25:
[Here is a Slack invite](https://join.slack.com/t/buildlog/shared_invite/enQtNDI2NDI4NDkxNTg4LWEzMTdlODU4ZDRmMjQ2ODU2NWM4MGFiZDZjM2NmN2YxYjgxZDQxOGM3YmEzNDNlOTgxNTZmMDVjOTAyMjg5ZmY). It expires soon.

hexxter, 2018-10-11:
the telnet feature is not working.
SDK:  v3.2-dev-39-gaaf12390
FW version:  1.1f (20180917) (ESP32)

if i set a telnet port and restart the port is after the reset 0 (in the config) and a portscan shows me only port 80, 81

BR Hexxter

luc-github, 2018-10-11:
Yes telnet bridge is not yet implemented. 
Front end is ready. I know it is confusing

closed 2019-10-14 by bdring