bdring/Grbl_Esp32 #36: Additional Button Actions

MRedmon, 2018-09-22:
I've got buttons hooked up to hold/start/reset but I've got an extra button.  I notice that you've made it easy to re-assign pins that aren't being used.  Thoughts on the best/easiest way to make a button feed the y-axis +1cm each time it's pressed (or something like that)?  

Thanks!

MRedmon, 2018-09-23:
Ah.  Thank you for your input.  That's probably a bit beyond me :-)  

One other quick question about buttons... you note that pull up resistors are required for the reset/hold/start buttons.  Is that accurate?  I only ask because there are a couple resistors already in place for each one and the there appears to be no similar requirement for the limit switch inputs.  The 5v supply for the sd card appears to have been fixed since some of your instructions were written so I didn't know if the pull up resistors also were added...

Thanks!

bdring, 2018-09-23:
If you have my board, the pull up resistors are in place. That note is more for people use Grbl_ESP32 without my hardware.

MRedmon, 2018-09-23:
Awesome!

closed 2019-10-14 by bdring