bdring/Grbl_Esp32 #39: Add Uploadable Machine Profiles

bdring, 2018-09-27:
It was suggested that we add a way to upload machine profiles. If you have any input, please reply on this issue.

There is a [public google doc](https://docs.google.com/document/d/1cd9bxBowu-AtklWsCPrqdVpWzPvXYFW9yVarjGJfPIA/edit?usp=sharing
) where I will organize the feedback.

msurguy, 2018-09-27:
I'm not sure if the existing doc is covering it, but basically the type of the tool is also something that can be set by some settings and be shared separately from the machine profile. 

Example scenario: perhaps I have a Coasty with a laser and somebody else has a coasty with a pen / marker driven by a servo. I want to share all settings with the other person, but omit the tool settings / type. 

The machine profile INI / JSON file sounds cool since it would be possible to create a web-based configurator/wizard that is simple to understand.

bdring, 2018-09-27:
@msurguy can you give an example of the actual settings your are talking about in your example?

INI/JSON - Yes I think a web configurator would be great, but I would like to see alternate ways too.

closed 2019-10-14 by bdring