bdring/Grbl_Esp32 #40: Re: Using the SD Card

terjeio, 2018-10-07:
I am currently working on a driver for a Texas Instruments LaunchPad and have recently managed to run grbl as FreeRTOS-task with ethernet streaming. I am now looking into adding SD card support, and will use the protocol extension you have described. 

I think using reset is a bit harsh in order to terminate a running job and see that you are proposing $FQ in order to avoid that. I reckon this implies that the "normal" input stream has to be read (and dispatched for processing) by the protocol primary loop when streaming from the SD card, but isn't that a bit dangerous? Wouldn't it be better to handle termination by issuing a top-bit set real-time command instead? This since such commands never reaches the primary loop, allowing the primary loop to be set to ignore any input not originating from the SD card.

terjeio, 2018-10-07:
Got my first job running off a card now and discovered another potential issue, the FILE and SIZE elements are separated by a comma which is a legal character in filenames, IMO it would be better to use the vertical bar instead which is not. Simpler to parse.

bdring, 2018-10-07:
We have not done anything with a quit feature yet.

Grbl reset "mc_reset()" is the normal way to stop a streaming job or in process move. The proper sequence is Feedhold to stop motion, then Grbl Reset. This will clear out the current move and the buffer, but not affect the machine position or state. Ideally, quitting an SD job would be similar.

We have been talking about implementing some "rules" in the code to avoid danger, but right now it is "trust the user". I agree that restricting access to the primary loop might be a good start.

I'll do some experiments using the reset method soon.

terjeio, 2018-10-07:
The vertical bar is used to separate the main elements in the realtime report, so there is a precedence for its use in Grbl. Should I raise an issue over at the master branch to adress it?

I am aware of the other characters reserved by Grbl and I filter out the files containing them. Presenting unusable filenames to the user is bad practice - or perhaps adding an optional third field to the report that indicates so is a better way? Just thinking aloud...

bdring, 2018-10-07:
The only thing right now that is affected is WebUI. We would have to coordinate that before a change.

luc-github, 2018-10-08:
The web ui does not rely on $ commands but direct SD API so changing $ commands output should not affect it

terjeio, 2018-10-08:
Is the web ui using the SD API for listing the files as well? If so, does that mean that no senders using the [FILE... report exists yet?

IMO to not confuse users, files with invalid names should be listed in the UI but flagged as such, thus avoiding questions about why they do not show up.

Valid files:

[FILE:/tesla.nc|SIZE:1023467]
[FILE:/tesla,file.nc|SIZE:1023467]

Invalid file:

[FILE:/tesla file.nc|SIZE:1023467|INVALID]

Next step for me now is to write an UI control in C# for handling the SD card.

luc-github, 2018-10-08:
Yes it was my meaning
Also in web ui, invalid names are displayed but print command is not available

luc-github, 2018-10-08:
IMHO instead of displaying  Invalid  the file should not be displayed at all

terjeio, 2018-10-08:
> IMHO instead of displaying Invalid the file should not be displayed at all

Adding the INVALID flag means that the decision to show them or not is up to the UI designer. I would prefer to show them, perhaps with a tooltip explaining why they are not usable.

luc-github, 2018-10-08:
In that case why not keeping same logic to all files and displaing all files present on SD ?

bdring, 2018-10-08:
I am **OK with changing to the vertical bar**.

I am not a fan of the INVALID labeling. I think it implies more than just file extension checking. Even files with a matching file extension could have internal reasons for being invalid.  I am **OK with displaying all files**.

This allows the sender to decide what types to show. If the sender wants to allow .luc files, there is probably a good reason for it.

BTW: I will work look into a good way to terminate an SD card job if the gcode is not valid in the file.

bdring, 2018-10-09:
- I have changed the directory listing to use the vertical line. WebUI seems to be OK with that.
- I have removed all file filtering. It is up to the sender to determine what to allow the user to do. 
- If you feed hold and then reset, the job will stop and close the file. The last line read will be reported.
- If your SD file has bad gcode, the job will stop and report the bad line number.

While everything seems to be OK, the WebUI is not displaying the error messages.

I will push the code to the WebUI branch soon.

bdring, 2018-10-10:
Sorry @luc-github I was wrong about that. It appears to work fine. The changes have been pushed to the WebUI branch

closed 2018-10-10 by bdring

bdring, 2018-10-11:
I was able to get the RMT to work without panics, but I was seeing inconsistent timing on the o'scope. The single step timing was fine, but time between steps was not as good. I also saw inconsistent timing with the direction pin.

I think there should be a way to make it work. I think the API might need to be looked at to optimize it for Grbl.

Please start a new issue on this, if you have more to share.