bdring/Grbl_Esp32 #41: pullup lim z not work

hexxter, 2018-10-10:
Hi,

i have the HW Version 2.1 and i have a problem with the limit switch Z. The Pullup does not work.
Is this a problem only for me or have other people the same problem?
For the next HW Version is a option for an externel pullup a good solution.

bdring, 2018-10-10:
That sounds like a problem with your ESP32.

hexxter, 2018-10-11:
i have fixed ist for me and use the mist pin for the z limit and no it works fine

gflaser-au, 2018-10-11:
@bdring For the next hw rev it could be a good idea to add some (bottom-side?) footprints for some extra pullup resistors. It could be helpful for those of us throwing a few tens of kilovolts around only a few inches away. (No-fit on your std hw build; just add the footprints)
Even for the Nano, most common recommended solution to false triggering hard limits is a more solid pullup than the chip provides (as well as RC filter network which looks to be already in place.) 

or optos  :)

bdring, 2018-10-11:
I will consider that for future revisions.

closed 2018-11-10 by bdring