bdring/Grbl_Esp32 #42: WebUI: jog feed rate limits

gflaser-au, 2018-10-11:
WebUI complains if I try to set a jog rate > 9999mm/min.
My laser machine is capable of much higher speeds (with a nano 20000mm/min, but likely higher with esp32)

WebUI should probably read grbl cfg items $110-$112 to set this upper limit dynamically. (LaserGRBL does this.)

or can the limit be increased to 100000mm/min as a temporary measure?

luc-github, 2018-10-11:
Yes this is possible - I will add this

luc-github, 2018-10-13:
I think removing the top limit check is the way to go, the  FW already have the limits defined  and handle it.
The original check was to avoid 0 as bottom limit, I forget why I have added a top limit check .....

luc-github, 2018-10-13:
@gflaser-au please give a try to https://github.com/luc-github/ESP3D-WEBUI/blob/2.1/index.html.gz

I have removed the top limit check in UI and preferences, and increased XY control feed rate size of 2 chars for your convenience
Hope is fit your needs

Sorry for the delay I am travelling and few time to code / test

luc-github, 2018-10-23:
@gflaser-au you may now close issue as you confirmed it is Ok

gflaser-au, 2018-10-23:
Tested and confirmed - using index.html.gz linked here.

closed 2018-10-23 by gflaser-au