bdring/Grbl_Esp32 #43: Missing functions for coreXY

cemir, 2018-10-15:
Hi,

while trying to compile with coreXY enabled, it apears that system_convert_corexy_to_y_axis_steps function is missing

STEPS :
uncomment :
#define COREXY

compile

regards

bdring, 2018-10-15:
Sorry. I am aware of that. That was missed when porting to ESP32.

I noticed that last week and added it back in an unpushed test branch. I will try to update the two public branches by mid week.

bdring, 2018-10-16:
I have added it to the WebUI branch, can you test that branch? If all goes well I'll add it to the master branch.

cemir, 2018-10-16:
Hi, thanks for your reactivity! unfortunately, I will not be able to test before tomorrow evening (UTC+2), I'll let you know

closed 2018-11-10 by bdring