bdring/Grbl_Esp32 #47: Hard limit causes crash when using SD Card

gflaser-au, 2018-10-28:
Hard limit ISR calls mc_reset, which calls file close if streaming gcode from an sd card file.
close attempts to take a lock which causes the crash (can't take a lock in interrupt context).

Unsure how best to attack this, but it would be nice to gracefully recover so a software reset could be issued to bring the machine back to operating state once the hard limit switch has been cleared.

Currently requires a hard reset / power cycle to recover.

gflaser-au, 2018-10-29:
PC: 0x400810c8: onStepperDriverTimer(void*) at C:\Users\V\AppData\Local\Temp\arduino_build_527734\sketch\stepper.cpp line 265
EXCVADDR: 0x00000011

Decoding stack results
0x4008fa24: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 140
0x4008fc27: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 149
0x40081725: lock_acquire_generic at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/newlib/locks.c line 141
0x40081851: _lock_acquire_recursive at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/newlib/locks.c line 169
0x400e5de1: VFSFileImpl::close() at C:\Users\V\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\libraries\FS\src\vfs_api.cpp line 288
0x400e5d4b: fs::File::close() at C:\Users\V\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\libraries\FS\src\FS.cpp line 136
0x400d3529: closeFile() at C:\Users\V\AppData\Local\Temp\arduino_build_527734\sketch\grbl_sd.cpp line 93
0x400d412b: mc_reset() at C:\Users\V\AppData\Local\Temp\arduino_build_527734\sketch\motion_control.cpp line 375
0x40080f69: isr_limit_switches() at C:\Users\V\AppData\Local\Temp\arduino_build_527734\sketch\limits.cpp line 53
0x40081261: __onPinInterrupt at C:\Users\V\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\cores\esp32\esp32-hal-gpio.c line 219
0x4008188b: _lock_release_recursive at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/newlib/locks.c line 215 (edited)

bdring, 2018-10-29:
I will think about this. I agree you should be able to recover from the keyboard.