bdring/Grbl_Esp32 #50: Add I2C I/O expander to card

tknapstad, 2018-11-07:
I would suggest adding an I2C I/O expander to the card to allow more flexibility and resolve the limitations when using an SD card etc. An example here from TI.

[TCA9555 16bit I2C I/O expander](http://www.ti.com/product/tca9555)

This adds 16 I/O lines using only 3 pins (SDA, SCL and an interrupt pin) on the ESP32 board that are 5V tolerant, ESD protected and provides open drain outputs. Perfect for limit switches, mist, probe, door etc. 

It only adds about a $1.50 to the BOM cost.

bdring, 2018-11-07:
I added that to the roadmap. Most of the I/O you are talking about uses interrupts. I see there is an interrupt pin option. I'll have to look at that more.

Do you have any suggestions on how to add that to cpu_map.h and config.h?

tknapstad, 2018-11-08:
Another option is this chip from Microchip which has two interrupt lines, one for each 8-pin port. Consumes one more ESP32 pin, but gives more flexibility:
https://www.microchip.com/wwwproducts/en/MCP23018

I haven't looked at the code until now. My initial impression and ideas:

- I don't think config.h needs to be touched
- Need to add an ioexp.h/.cpp module to handle the IO expander interrupt(s) and I2C communication. 
- Pin numbers can be defined in ioexp.h. The upper bit can be set to 1 for the IO expander pins, so we start at 127 (#define GIPO_EXP_NUM_1 127)
- Include ioexp.h in cpu_map.h so that we can assign the IO expander pins to functions like PROBE_PIN
- An abstraction level is needed for at least the Arduino functions pinMode, digitalRead, digitalWrite and attachInterrupt so that we can handle both regular pins and the IO expander pins in the same way from the Grbl code that
- The IO expander interrupt handler must read the state of all pins and call the appropriate existing interrupt handlers (for instance isr_limit_switches if any of the limit switches were triggered)

The most work seems to be in creating an abstraction layer for GPIO handling (pinMode etc.) and implementing the I2C protocol for the IO expander.

npross, 2019-01-23:
I've never used it but I was looking for I2C IO expanders and came across these:
https://www.semtech.com/uploads/documents/sx150x_89b.pdf

Digikey sells them for $2.55 
They are QFN

Sparkfun has a breakout board for and there is code too.

MakersHeaven, 2019-02-05:
How about the cheap PCF8575 as a port expander? There are plenty of chips and modules out there and it's almost gold standard in ways of simplicity.

svalambhia, 2020-04-09:
This guy uses esp8266 and shift registers to control GRBL 8-axis. https://github.com/gcobos/grblesp & youtube video https://www.youtube.com/watch?v=EGESCcurkwA For a Good controller we need 9 axis like g2core https://github.com/synthetos/g2