bdring/Grbl_Esp32 #51: Add Feedback to Override Feed Rate

negulescus, 2018-11-08:
Hello,

I plan to use this grbl port in a Wire-Cut EDM ([https://en.wikipedia.org/wiki/Electrical_discharge_machining](https://en.wikipedia.org/wiki/Electrical_discharge_machining)).
[image]

Simply put, Wire-Cut EDM, is a CNC with two axis and a molybdenum or brass wire on which a voltage is applied. Between the wire and the metallic work piece a discharge happens (usually through a dielectric), when the distance between them is adequate. This distance (gap) determines the voltage and the current through the discharge circuit. This current must be kept constant by not allowing the wire to touch the work-piece and by keeping the gap constant.

_**The value of the current through the circuit would be the feedback needed to override the feed rate.**_
This would result in a feedback loop that would not allow grbl to increase the speed of _x_ and _y_ motors if this would result in a smaller gap, thus resulting a bigger current (worst case scenario = short circuit).
I believe a PID would be necessary too (not 100% sure, as these systems do not cut wit more than a couple of mm per minute).

I believe that an external circuit that would read the current value (average it over a few milliseconds) and outputting a proportional analog value that would be read using an analog input would work.
Another solution would be to use a digital input pin that would be toggled by this circuit slower or faster depending on the value of the current.

If anybody has another ideas how this could be better implemented please comment.

misan, 2018-11-08:
@negulescus your assumption is that GRBL should control the loop. An alternative idea that does not require to modify GRBL is to use it as is and handle feedback outside of GRBL code. Let us say software commands tiny movements every now and then, based on detecting an increased gap (measured by an external device/circuit), whenever current goes below a threshold while a retract motion could be sent when a too high current is detected (short).  That said, a grbl fork for EDM could make perfect sense too.

bdring, 2018-11-08:
I am not an expert in PID, but I am concerned about it having enough range to work with. In an email you gave me some of the parameters.

"Speeds of between 5 and 20mm/minute are sufficient. The resolution of 100 steps/mm would be more than enough."

At 5mm/min and 100 steps/mm you have a speed of 8.3 steps/sec. Regardless of your control loop speed, you are only taking steps at 8Hz.

Do you think that will work? I would think you something faster so the PID can run faster and have a reasonable range to work with.

jeffeb3, 2018-11-14:
If you make the control loop outside grbl, is there a way to reduce the buffer? Feedback control loops don't work well if their output goes into a buffer.

bdring, 2018-11-14:
Feedrate overrides are not buffered, they are acted upon "immediately". But that depends on the step rate. If the step rate is very slow, I think the update rate would be slow.

negulescus, 2018-11-16:
Hello everybody and thank you for your input!

@misan external control of the loop might also work but I see it as less ideal. Preferably everything would be kept as small a possible and with minimal changes, within grbl.

@bdring EDM is indeed a slow process. Calculating, at min and max cutting speeds results a frequency between 8 and 34 Hz.

I am not sure if PID is mandatory for this feed rate feedback override, but I know PID can be also used for slow processes (e.g. temperature control - http://www.topcontrol.com/fichiers/en/FastMethodsForSlowLoops.pdf).

Regardless if PID will be used or not, I would just need grbl to act like this (proposal):

- read a global variable of type unsigned char (byte) [0:255] (which would be changed inside an interrupt which reads an analog pin - hopefully this wouldn't be too slow)
- use that value to override the feed rate (e.g. for the command G1 X30 Y90 Z55 F20, the actual feed rate would be 20*value/255)

bdring, 2018-11-16:
I might write some test code. If it works, I will publish it, but I doubt it will enter the master branch until it proves to be useful.

Grbl has a built in feed rate override that I will use. You set a target feed rate in your gcode and the override will adjust that +/100%. I am worried there will be noise on the analog line that causes the feed rate to  constantly adjust a tiny bit. I may need to filter noise. It adds work to the planner each time it changes.

negulescus, 2018-11-17:
That would wonderful! Thank you!

svalambhia, 2020-04-09:
Hi Bart,

**Feed override** on you design,  https://www.cnc4fun.com/it/home-di-prova/ has implemented feed override on your design. He had also released design files, bit difficult to find on Italian website. I think this is must have option for practical use.