bdring/Grbl_Esp32 #56: error while compiling

cnc4less, 2018-11-18:
i am using winxp10/64bit with arduino 185 , when i compile i get this error
exit status 1
'client' may be used uninitialized in this function [-Werror=maybe-uninitialized]

luc-github, 2018-11-18:
in what file this happen ?

cnc4less, 2018-11-18:
comes after 
Using library ArduinoOTA at version 1.0 in folder: C:\arduino-1.8.5\hardware\espressif\esp32\libraries\ArduinoOTA 
exit status 1
'client' may be used uninitialized in this function [-Werror=maybe-uninitialized]

cnc4less, 2018-11-18:
[error.txt](https://github.com/bdring/Grbl_Esp32/files/2592617/error.txt)

luc-github, 2018-11-18:
In that case it is in core file not GRBL code - so the current solution is to disable this warnings in preferences:
[image]

luc-github, 2018-11-18:
Ok reading at your error.txt seems indeed in some GRBL code - until it is fixed you can disable warning

cnc4less, 2018-11-18:
just did and got another error
Using library ArduinoOTA at version 1.0 in folder: C:\arduino-1.8.5\hardware\espressif\esp32\libraries\ArduinoOTA 
exit status 1
'class WiFiClass' has no member named 'getSleep'

luc-github, 2018-11-18:
how did you installed ESP32 core ? this is not the right location

cnc4less, 2018-11-18:
C:\arduino-1.8.5\hardware\espressif\esp32

cnc4less, 2018-11-18:
[image]

luc-github, 2018-11-18:
Where you put is not usual place so your installation may be wrong - that is why I have asked how did you installed - there may be some issue with your way
Normaly you use preferences in arduino and boot manager as mentioned in ESP32 github

cnc4less, 2018-11-18:
C:\arduino-1.8.5\hardware\espressif\esp32\libraries\WiFi\src\WiFiAP.cpp:40:43: fatal error: dhcpserver/dhcpserver_options.h: No such file or directory

compilation terminated.

luc-github, 2018-11-18:
Yes installation is not correct - they are all core files called by libraries present in esp32 core - so there is issue with path / installation

gflaser-au, 2018-11-18:
Did you try to compile any esp32 examples?
I followed the instructions recently - followed exactly and had no problems compiling grbl_esp32.

cnc4less, 2018-11-18:
i will install arduino 187 start fresh for all and see what happens

luc-github, 2018-11-18:
I use 1.8.5 without issue on my side - I have reproduced the [-Werror=maybe-uninitialized] and working on fix the ones related to GRBL - unfortunaly still have some in core that need to be fixed - so keep Warning to none or default in your fresh install

luc-github, 2018-11-18:
once you are done with your fresh installtion try to compile any example from esp32 repository - if compilation is Ok you should be good for GRBL_ESP32

cnc4less, 2018-11-18:
fresh installation with arduino 187 compiled fine. thank you.

luc-github, 2018-11-18:
Great so I guess issue can be closed

closed 2018-11-18 by bdring

koka7777745, 2019-07-25:
does not compile, gives an error
[image]