bdring/Grbl_Esp32 #60: Support RFC2217

Harvie, 2018-11-24:
rfc2217 is telnet extension for acessing remote serial line. Maybe it would be usefull to implement, because there are drivers (for most operating systems) to attach such terminal server as local serial port, so it can be used by any software which supports local serial ports.

https://tools.ietf.org/html/rfc2217

luc-github, 2018-11-25:
Do you have example for such drivers ?

Harvie, 2018-11-25:
I've done some more research today and i've found they are probably not really drivers. They are just userspace programs, that somehow manage to add new fake device. I've written all my observations here:

https://github.com/vlachoudis/bCNC/wiki/Remote-Machine

luc-github, 2018-11-25:
FYI :
* TCPCOM32 https://sourceforge.net/projects/tcpcom32/ is a very old project that work on WinXP 32 bits only, I remember someone did a version for win7,  but it was not really working ... 
* I cannot test VPSE http://www.eterlogic.com/Products.VSPE.html as free version is only for 32bits and when I see that their 64 bit reference is `Vista 64`,  I won't take risk to buy a license.
* https://www.hw-group.com/product-version/hw-vsp3-single seems working with current implementation of Telnet in ESP3D and GRBL_ESP32

luc-github, 2018-11-25:
Sorry to ask, I am not expert so what the benefit of rfc2217 for GRBL_ESP32 vs RawTCP ? 

GRBL_ESP32 only read and write data, so RawTCP should be enough, but I may be wrong.

@bdring : Does GRBL_ESP32 could handle control lines, what would the benefit if yes ?

bdring, 2018-11-25:
What control lines? Hardware flow control?

luc-github, 2018-11-25:
control lines are RTS / CTS /DTR /DSR,  so yes hardware flow control I think

luc-github, 2018-11-25:
Ok understood - thank you

Harvie, 2018-11-25:
> The USB-serial on dev boards is not wired that way.

I think that telnet "serial" interface of esp32 grbl is not wired at all, so settings of baudrate and flow control might be completely irrelevant. I've done some research and i think the virtual serial might work out of the box.

Is there somebody, who can try it? I don't have windows to try it.

On linux it should be easy:

`socat pty,link=/dev/ttyS23 tcp:hostname:port`

Harvie, 2018-11-26:
BTW how am i supposed to reset ESP32 Controller. Currently in bCNC we reset arduino by pulsing DTR pin. I need some alternative for this in order to add telnet support to bCNC.

bdring, 2018-11-26:
The esptool.py program that is used to program ESP32 dev boards via bootloader uses DTR and RTS.  See this piece of the schematic on how to do that.

[image]

There is also a [ESPxxx] command to do it. 

**[ESP444]RESTART pwd=<admin/password>**

Harvie, 2018-11-26:
> See this piece of the schematic...

You need to differentiate between serial and telnet. There are no serial wires when running over IP.

bdring, 2018-11-26:
[ESP444]RESTART pwd=<admin/password>

bdring, 2018-11-26:
What is the purpose of the reset? Do you just need to reset Grbl or the ESP32? You will lose your Telnet connection if you do that. 

Would you like to join our Slack channel?

Harvie, 2018-11-26:
Talking about arduino: In bCNC we have two buttons "soft reset" (ctrl+x) and "hard reset" (DTR pulse).
When something gets screwed and i get to the "Alarm" state it needs soft reset. But sometimes stuff freezes even more and i need to do hard reset.

I suppose there will be similar issue with ESP32. I plan to implement serial over TCP support to bCNC relatively soon, but i don't know how to handle hard reset button in such case. In fact i don't even know if Ctrl+x will result in soft reset. I will need to experiment with combination of ESP32 GRBL telnet and bCNC.

Another thing that worries me is login to that telnet service. This wasn't really used on plain serial connection, so i will need to handle that. I wonder if there is some standart login protocol for telnet, so this will not require us to have separate code for ESP32 GRBL and eg. ESP32 smoothie or tiny-g (if they ever decide to implement their own telnet). Because bCNC supports more controllers than just GRBL. Maybe we should contact authors of other motion controllers (tiny-g, smoothie, marlin, etc...) to agree on some connection and auth mechanisms that will be compatible between controllers. Maybe it would even make sense to support some basic SSH connection which would handle auth stuff for us. AFAIK there are SSH implementations running on ESP32...

> Would you like to join our Slack channel?

I don't have slack account. In fact i don't even really know what slack is :-) But surely i would like to be in contact with your team regarding bCNC and GRBL-ESP32 cooperation. I still have some major task to solve in bCNC before i start working on telnet support. I've just discovered that GRBL-ESP32 exists, so i wanted to start the discussion. There is surely some work to be done in the future.

bdring, 2018-11-26:
Slack is designed for groups/teams to work in real time. It has some integration with GitHub. [Try the link](https://join.slack.com/t/buildlog/shared_invite/enQtNDg0ODA4MjcyOTE0LTg3YWNkYmQxNzkwZGYxMDBiZTIwOGI5OGY4ZmNhNjNlZWZjZDJhNTRhYTg2ZWRmODhlMGNlNDBiYjhjOGEzZTM) if you want.  It is pretty easy to use. The link expires tomorrow,

Harvie, 2018-12-01:
Sorry for not responding, i am bit busy lately. However i have some good news.

I've just found that serial library used in bCNC already has abstraction layer with native IP support (Both RFC2217 and plain TCP). I've written some details here: https://github.com/vlachoudis/bCNC/issues/1072

So that's good starting point. You can try to connect bCNC to ESP32.

Harvie, 2019-03-03:
I've proposed simple auth protocol for networked GRBL and i can add such support to bCNC. So i want to ask you for your opinions:

https://github.com/vlachoudis/bCNC/issues/1072#issuecomment-469010560

thawkins, 2019-10-26:
Socat on linux is able to bind a virtual tty (/dev/ttyxxxxx) to a port on a remote system, 

There is also another project that purport to be abke to do this 

https://gitlab.com/lars-thrane-as/ttynvt