bdring/Grbl_Esp32 #63: Raspberry PI Hat version

edcasati, 2018-12-01:
I am really impressed by this whole project. Many Kudos to the author.

Just a heads-up in case anyone is interested.
I am working on a variation, for personal use, that will plug directly in to a Raspberry Pi. 
I could use a standard Arduino, but I love the ESP32 for all things
It is mostly a personal challenge for myself, but I have other good reasons:
a) There are many mature software packages such as CNCJS that run well on a Raspberry Pi.
b) You can plug in a monitor, keyboard and mouse.
c) Support for thumb drives.
d) Frees the ESP32 to run GRBL without sharing tasks
as a side note, the CNC that I am building needs drivers more powerful than the on-board kind, so hookup is to external drivers.

Here is my progress so far,, utterly untested and unfinished:

[PCB_ESP32-Wroom-to-Pi-Board_20181130205111.pdf](https://github.com/bdring/Grbl_Esp32/files/2635566/PCB_ESP32-Wroom-to-Pi-Board_20181130205111.pdf)
[Schematic_ESP32-GRBL-Raspberry-PI-cape_Sheet-1_20181130205219.pdf](https://github.com/bdring/Grbl_Esp32/files/2635567/Schematic_ESP32-GRBL-Raspberry-PI-cape_Sheet-1_20181130205219.pdf)

gflaser-au, 2018-12-01:
I took a quick look at your schematic and have a couple of comments.
1, r6 and r1 look incorrect.  I think you intend those to be pull-ups, they should connect between 3v and the inputs ( not across the switches)
2, the limit switch capacitor pack u7 should be on the other side of the 100ohm resistors. Also consider adding footprints for more pull-up resistors on the limit sw inputs ( connected to the esp pins.) additional pull-up has helps lots of people with noisy setups. These inputs are very sensitive.
3, as matter of convention, gnd symbols “point” down. ;) 

I’d urge you to test out the esp32 WebUI. It is actually very useful and you can then connect directly from and pc / laptop via WiFi to upload g code files to the sd card and control the machine.
The only thing missing from the likes of CNC.js and (my fav) lasergrbl, is the visualizer.

gflaser-au, 2018-12-01:
Oh and, if you have an esp32 nodemcu board that you can program - you can test out the WebUI without actually having a CNC machine.  Just squirt in the code and make sure grbl
Is configured to not use hard limits or homing - and either pull up the 4x control inputs or comment them out in cpu_map.h.
From there you could attach an sd card to the right pins and upload file if you wanted to. It’s worth testing out for sure. 😁

edcasati, 2018-12-01:
Thank you so much for your feedback.
As you can see, I really don't know much of what I'm doing.  I recently retired for health reasons, and aside from needing to find a hobby or drive my wife crazy, I need too really force my brain to keep it working. Hence the challenge like this one. But enough of that...

Is there a mistake in your posted V3.1 schematics? It looks like the 'Hold' input is miswired. 
http://www.buildlog.net/blog/wp-content/uploads/2018/10/schm_esp32_cnc_test_v3p1.pdf

I made every change you pointed out or recommended. Now to double check things and the part that I'm dreading... routing the wires.

closed 2018-12-04 by edcasati