bdring/Grbl_Esp32 #65: CO2 Laser controll

pityuka3d, 2018-12-04:
I like your work and I want to put a CO2 laser cut with this control instead of Arduino Mega.
I made a test bench. I set the laser mode and what I found information. But while the gcode is running, the laser does not turn on. I measured and output at 0.15 volts at 80%.
If I run the command directly (gcode sender) then it works fine.

Is there any way to attach a nextion display? I would like to experiment with it.

Thanks
I'm sorry but my English is not good.

bdring, 2018-12-05:
Are you checking the PWM with a scope or logic analyzer? You may need to shift the voltage from 3.3v to 5v.

There are no plans to add a display. A phone or web browser is the plan right now.

With that said, I don't think it would hard to add a display.

pityuka3d, 2018-12-05:
Thanks for the reply!
I'm sure you have to switch to 3.3 volts. Because the power supply of the laser is 5, it works with PWM. That's why I thought I'd try a smooth Led.
Interestingly, it just does not work properly during work. Test operation or direct command works well. What do you suggest to change?

bdring, 2018-12-05:
@gflaser-au has had good success with his CO2 laser. He made some changes to the PWM for best performance. I don't recall the exact changes, but he did change to 5V via a level changer.

pityuka3d, 2018-12-05:
Thank you very much for your help!
Unfortunately, it's 2 o'clock here, I do not have a lot of options.
I used the NPN transistor to solve the shift. You are right with the PWM settings. Now you can turn the laser on every program. But you can not go over 80%. If I set this value above, do not turn the laser on. Or I can not go under 2. (with so many S1 commands)

bdring, 2018-12-05:
with max RPM set to 1000 here is S950 (95%)

[image]

With S10 (1%)

[image]

These are my settings

```
#ifdef CPU_MAP_ESP32
	// This is the CPU Map for the ESP32 CNC Controller R2	
	
	  // It is OK to comment out any step and direction pins. This
    // won't affect operation except that there will be no output
		// form the pins. Grbl will virtually move the axis. This could 
		// be handy if you are using a servo, etc. for another axis.
		#define X_STEP_PIN      GPIO_NUM_12
		#define Y_STEP_PIN      GPIO_NUM_14
    #define Z_STEP_PIN      GPIO_NUM_27		
		
		#define X_DIRECTION_PIN   GPIO_NUM_26
		#define Y_DIRECTION_PIN   GPIO_NUM_25  
		#define Z_DIRECTION_PIN   GPIO_NUM_33 
		
		// OK to comment out to use pin for other features
		#define STEPPERS_DISABLE_PIN GPIO_NUM_13		
		
		// *** the flood coolant feature code is activated by defining this pins
		// *** Comment it out to use the pin for other features
		#define COOLANT_FLOOD_PIN 	GPIO_NUM_16			
		//#define COOLANT_MIST_PIN   	GPIO_NUM_21
		
		// If SPINDLE_PWM_PIN is commented out, this frees up the pin, but Grbl will still
		// use a virtual spindle. Do not comment out the other parameters for the spindle.
		#define SPINDLE_PWM_PIN    GPIO_NUM_17 
		#define SPINDLE_PWM_CHANNEL 0
		// PWM Generator is based on 80,000,000 Hz counter
		// Therefor the freq determines the resolution
		// 80,000,000 / freq = max resolution
		// For 5000 that is 80,000,000 / 5000 = 16000 
		// round down to nearest bit count for SPINDLE_PWM_MAX_VALUE = 13bits (8192)
		#define SPINDLE_PWM_BASE_FREQ 5000 // Hz
		#define SPINDLE_PWM_BIT_PRECISION 12   // be sure to match this with SPINDLE_PWM_MAX_VALUE
		#define SPINDLE_PWM_OFF_VALUE     0
		#define SPINDLE_PWM_MAX_VALUE     4096 // (2^SPINDLE_PWM_BIT_PRECISION)
		
		#ifndef SPINDLE_PWM_MIN_VALUE
				#define SPINDLE_PWM_MIN_VALUE   1   // Must be greater than zero.
		#endif
		
		#define SPINDLE_PWM_RANGE         (SPINDLE_PWM_MAX_VALUE-SPINDLE_PWM_MIN_VALUE)		
		
		// if these spindle function pins are defined, they will be activated in the code
		// comment them out to use the pins for other functions
		//#define SPINDLE_ENABLE_PIN	GPIO_NUM_16
		//#define SPINDLE_DIR_PIN			GPIO_NUM_16		
		
		#define X_LIMIT_PIN      	GPIO_NUM_2  
		#define Y_LIMIT_PIN      	GPIO_NUM_4  
		#define Z_LIMIT_PIN     	GPIO_NUM_15 	
		#define LIMIT_MASK      	B11
		
		#define PROBE_PIN       	GPIO_NUM_32  
		
		#define CONTROL_SAFETY_DOOR_PIN   GPIO_NUM_35  // needs external pullup
		#define CONTROL_RESET_PIN         GPIO_NUM_34  // needs external pullup
		#define CONTROL_FEED_HOLD_PIN     GPIO_NUM_36  // needs external pullup 
		#define CONTROL_CYCLE_START_PIN   GPIO_NUM_39  // needs external pullup    		
		
#endif```

gflaser-au, 2018-12-05:
I can post my settings later, however:
1, the npn will invert the signal, unless you used two.
2, most laser power supplies need pwm frequency above 20kHz.
3, lasers will not fire below about 10% pwm. To fix this grbl has a minimum pwm setting. ( you can see my setting when I post later)

4, instead of npn, I used a high speed optocoupler, to isolate esp32 from laser psu.

Esp32 grbl is very good for a Co2 laser. Mine 100w and I did have some problems with noise interference. I have solved these problems with a combination of hardware and firmware changes and not now works great!

pityuka3d, 2018-12-05:
I tried to connect a negative with a transistor. And I tried the positive with 2 transistors. But no sign was ever coming out. While Gcode is running, there is hardly anything. Now I'm going to see that almost constant earthing is signaling the output. So you can not honey with it. But the diameter of a positive fiber is bought by 3v.
Unfortunately, I can not measure the scope because it is broken.
I think it would be very good for this control too. So far I did Arduino Uno but I did not like it so I switched it to Arduino Mega. It would be great to have this control. I want to do it. But I got jammed. :-(

gflaser-au, 2018-12-06:
I recommend typing in some g code commands to the sender. Then you can test easier. (I use LaserGRBL)
(After homing, or setting zero position)
manually enter in:
M3S500
G1x1y1

if max speed is 1000, you will see 50% PWM output - provided you do not use any other commands that will cause laser to turn off. (G0 for example). 
Then you can use:
M3S250 to set 25%, M3S750 to set 75% etc.

If you do not have a 'scope, it will be difficult - but try with a multimeter on DC and check the averaged DC value (and frequency if your meter has it.)
Try with no transistor connected, on the ESP32 PWM pin. Then continue from there.

I am happy to check your design. Can you post a schematic diagram of your circuit with the NPN transistors and all part names and resistor values?

remember to set:
		#define SPINDLE_PWM_BASE_FREQ 20000 // Hz
to get 20kHz.

bdring, 2018-12-06:
@gflaser-au 

Can you provide all of your spindle #defines?

gflaser-au, 2018-12-06:
I can - just not immediately. I'm not close to my home office for a bit.
It's be great if we could figure out a way of adding machine specific cfg files to the repo ;)

As a starting point, the defaults + 20kHz should be ok.
I kept my setup at 8 bit resolution, but the defaults should be fine to testing the hardware setup.

bdring, 2018-12-06:
I have added a few extra machines to the master branch cpu_map.h.  Look how I did it and see if you can add your machine.

pityuka3d, 2018-12-06:
I am using laserWeb (and other) Unfortunately, the instructions you entered do not turn on the laser. Only for program-controlled control. Now I have a multimeter. Tomorrow I'm trying to get a scopot. The output of the board (17pin) is 3.3m:
Feed 0, S 1000 = 2.31v output
feed 1000, S 1000 = 0.0v output
Feed 1000, S500 = 0.8v output
Feed 0, S500 = 1.15 outputs
Feed 10, S1000 = 0.15v output
Feed 10, S500 = 0.0v output
It does not respond to the entered M3 instruction in any program. (prints "OK") There are no problems with the motion commands.
Tomorrow I will test some transistor settings and draw the drawings.
I look at the file cpu_map.h. Based on your schematic drawing I knit all legs. But for testing, I'm not doing PCBs.
With the exception of the proposed 20,000 Hrz, everything is in the factory #defines.

gflaser-au, 2018-12-06:
M3 just sets spindle control state to on. S sets speed It will not move any axis.
Did you enable laser control in the grbl cfg?
If you haven't made any g1 moves after reset, use 
G1x1y1F1000 
(Feed rate will need to be set but you should see a warning if you missed that.)

In laser mode, G1 cmd is needed to actually make the pwm start - as a safety feature, the laser pwm disables unless in motion. G0 will force PWM = 0, even if still has M3S500 set.

pityuka3d, 2018-12-06:
Laser control is enabled. How does the M3 command work?
But PWM's strength is not good !!!
I'm trying to make it simpler.

If the axes move, they take the value off. But in the upright position it has only 2.15 volts of 3.3 volts.

I turned off the safety minimum movement. This is how the M3 command works.
But LaserWeb has been able to turn the laser on Testing so far. Even if the G0

Update:
I removed all hardware from the D17. If the axes do not move, they have a 3.3volt at 100% (M3S1000) and 1.64volt at 50% (M3S500).
If any axis moves (example: M3S500 G1x1y1) the value drops to 0.15 volts or below!

I'm sorry but my English is still not good!

pityuka3d, 2018-12-06:
I came across an Arduino Mega with GRBL. Basically, it also works with 5 V PWM. So there is no need to raise the level, but that was not the point.
4.75 in the case of 100%
50% was 2.41
Shafts are identical in motion and without motion.

But there is no stoppage monitoring feature here, and it's far behind the firmware you make!

gflaser-au, 2018-12-06:
Did you read this page? 
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Laser-Mode

It explains laser mode for grbl, and how M3 / M4 works.

Your description of pwm at s1000 / s500 is ok.
I don’t know why it would be changing when axis moves ( g1 commands )

Draw your circuit and upload it.  We will check it for you.

Also, make sure your power supply is powerful enough to run the stepper motors and the controller with some extra capacity. I use 10A 12V for 2 nmea 17 steppers (~ 2A rating)

pityuka3d, 2018-12-06:
Most I read it here relevant m3 the description of M4. This the third laser cutter what I build up. Although it is never M4 for me was going adequately, it M3-at I use it. There is not an electronic circuit yet currently because I cleansed it totally, that not disturb nothing! 

There are a couple of pictures here from the problem:
if the axes move takes away PWM performance.
(I cannot write it down more simply)
[image]
[image]
[image]
[image]

PWM level earned one making a step with the net and I tried some. Here an example:
https://www.incrediblediy.com/2013/02/simple-33v5v-level-shifter.html
https://www.google.hu/search?q=3v3+pwm+to+5v&source=lnms&tbm=isch&sa=X&ved=0ahUKEwjohaefuovfAhWNyqQKHWR-A20Q_AUIDigB&biw=1931&bih=1267

I tried a lot, but the 3.3voltos there is a problem with a sign.
How you would tie it up and with what kind of settings? Sends and I try it.

pityuka3d, 2018-12-06:
I forgot that 12v 10A I use nutriment 2 possibly 3 Nema with 17 engines (one by one 1.7 Ampere)
The pumps from a separate power source I operate it.
should give only an output sign now, or to control one led.
I do not understand it how does this why. I rearranged it onto an other output the cpu_map.h ban. And there this phenomenon. I gave a thought to it already that wrong the board

pityuka3d, 2018-12-06:
This helps possibly:
[image]
I know that this instrument there is a game but this now.
The controller board nothing is tied up now, only the scope

misan, 2018-12-06:
This never goes to +5V. It looks as if it needs a pull-up resistor.

However, when you showed the measurements with your meter voltage was
significantly higher that want can be seen in the scope, which makes me not
to trust at all these last measurements. Amazingly however, it shows a
plausible duty cycle.

pityuka3d, 2018-12-06:
Can not I get this error message to the gcode containing the M4 instruction?

error: 20 - Unsupported or invalid g-code command found in block.

There is no such error message with M3.
GRBL_Mega works well with M4.

pityuka3d, 2018-12-07:
There is a difference, but both of them should work. M3 is a continuous laser performance and M4 is dynamically changing. One is better for the diode laser than the other for the CO2 laser. Here, however, it would link the firmware to the gcode movement with the power of PWM.
Let me know how to log in.
I try every idea! :-D Or where to find the fault?
If you also set an esp board, is this the bug for a laser?

bdring, 2018-12-07:
I have never used M4 mode for lasers. I think the code is in ESP32 version but disabled because of the pin definition. I can check into in a few days.

I can verify PWM stays on during moves in laser mode.  

I don't understand why your PWM looks like a sine wave rather than square wave. Is there some capacitance or load on the I/O.

gflaser-au, 2018-12-07:
M4 works - I use it when engraving. @bdring the grbl laser mode page has a description. "Dynamic power mode. - scales power during accel / decel.

@pityuka3d Looks to me like your scope is AC coupled - or is connected via a capacitor to the PWM. Try DC Coupled, or another scope it you can get access to one.

Also, don't use S1000 to test... use S250 / S750 first (to check PWM polarity is correct) and them move out to S100 / S900. 
S1000 will give 100% PWM = DC ;)

Also, a feed rate F10 is very slow (I assume you are using metric settings?). Try S1000 or higher, but remember that any G0 (rapid) moves in your test gcode will turn the laser OFF - so if the gcode has a mix of G1 and G0, you will see lower DC voltage.

You should make a g-code file manually in a text editor.  Only use M3 then G1 commands.

I have tested exactly like you (esp32 only) and could easily see the correct DC voltages on PWM pin.

bdring, 2018-12-07:
I am also seeing similar issues. The PWM is turning off when a move is made. When LaserGrbl is sending the gcode with lots of Sxxx values the PWM stays on, but simple gcode is having a problem.

This must have made it into the code recently. I'll try to figure out what happened.

bdring, 2018-12-07:
I think I may have found the problem. I recently made a change to increase the PWM resolution to 12 bit. The PWM calculations in some places is a uint8_t.

@pityuka3d in cpu_map.h try changing to these values and recompile.

```
#define SPINDLE_PWM_BIT_PRECISION 8
#define SPINDLE_PWM_OFF_VALUE     0
#define SPINDLE_PWM_MAX_VALUE     256
```

pityuka3d, 2018-12-07:
@bdring 
It seems that this solved the problem. THANK YOU VERY  VERY VERY VERY MUCH!!!!!!!!!
The PWM sine signal is due to the wrong instrument. These Chinese "do yourself" devices are not something reliable. I used M4 when I had to cut a material that melted or engraved if I did not want it to be bent at the edge of the direction change.
I continue to test and share every experience.

@gflaser-au 
We still do not understand each other. Probably my bad English knowledge and Google Translator is the reason!
The scope is in DC mode and there is nothing connected to the PWM signal! This instrument is so bad.
The PWM signal was not stabíl. This can even be seen on a simple LED.
Feed rate and PWM intensity are irrelevant. I testing S100 to S1000 and F10 to F5000. Whatever 
It is still the end result. Only those pictures were uploaded. It seems that we are not using the same FW version.
For example, you wrote that you have M4 on it. This error did not happen to you, but at bdring I did. Can you send the FW version you are using?
After the changes suggested by Bdring, I already get that signal while I'm moving. If there is a problem then it is sure to be a Gcode error.

bdring, 2018-12-07:
@pityuka3d glad to hear it fixed your problem. I will push the fix soon. I may look at the M4 issue over the weekend. @gflaser-au is using the 20kHz which limits the resolution, so I think he was using 8 bit resolution.

pityuka3d, 2018-12-07:
I took the cpu_map.h file in
#define SPINDLE_DIR_PIN GPIO_NUM_16 
banned and goes to M4.
If I remember the foot of the floor, I'll look at it. Now the M4 will also turn the laser on properly, but then I'll chime and engrave with it a bit and you'll find out .....
The 20 Khz is a standard frequency for CO2 lasers. Generally, each power supply requires the PWM signal at this frequency or higher.
I'm doing a slice to make the laser unsuitable and put the 5v level up. Also, I release the free legs for further development (lcd, controller, etc.).
If I'm interested, I'll send it if I'm ready.

gflaser-au, 2018-12-08:
Sounds like @bdring has it solved.  I did not see this problem because rim using 8 bit resolution.
My version is very close to latest from WebUI branch.

That waveform on your scope looks exactly like ac coupled, so maybe it is broken. Try to check with another scope to verify it.

I stayed with 8 bit mostly because it had carefully tuned the min pwm count before you added higher resolution.... which of course changes min value.... and 8 bit is enough resolution. This means I didn’t see the bug @bdring has now fixed.

pityuka3d, 2018-12-08:
The problem was resolved. I am very grateful for the help.
I think it's pretty much the 8-bit resolution to control a CO2 Laser.
Unfortunately, this Chinese DSO138 scope has succeeded in getting it fast so much. Sometimes it even shows its own reference signal to sine. :-D
The optic coupler is a very good idea and disconnects the ground from the control. I have not thought about this yet. I'm trying. So far, this control likes most of all I've tried.

bdring, 2018-12-08:
I [updated](https://github.com/bdring/Grbl_Esp32/commit/eaad780820a0132aef87a02b0ca5fc82810a4b64) the master branch to fix the M4 command. 

M4 is now supported if SPINDLE_DIR_PIN is defined or laser mode is on.

I am going to close this issue. If new problems come up, please start a new issue.

closed 2018-12-08 by bdring

pityuka3d, 2018-12-12:
Fantastic! Thank you very much!!!