bdring/Grbl_Esp32 #66: Issue during compilation

moromusa, 2018-12-05:
Hello,
I am trying to upload the Web_UI, and I have an issue with the BTconfig.cpp during compilation. I tried with Arduino IDE versions 1.8.5, 1.8.6 and 1.8.7. Also with Partition Scheme by default, minimal and minimal SPIFFS (modified). Always with the same result.
I am probably making some mistake so help would be great. Thanks.

Here I copy the error message:

Arduino:1.8.7 (Windows 10), Tarjeta:"ESP32 Dev Module, Disabled, Minimal SPIFFS (Large APPS with OTA), QIO, 80MHz, 4MB (32Mb), 921600, None"

BTconfig.cpp:52:23: error: variable or field 'my_spp_cb' declared void

 static void my_spp_cb(esp_spp_cb_event_t event, esp_spp_cb_param_t *param)

                       ^

BTconfig.cpp:52:23: error: 'esp_spp_cb_event_t' was not declared in this scope

BTconfig.cpp:52:49: error: 'esp_spp_cb_param_t' was not declared in this scope

 static void my_spp_cb(esp_spp_cb_event_t event, esp_spp_cb_param_t *param)

                                                 ^

BTconfig.cpp:52:69: error: 'param' was not declared in this scope

 static void my_spp_cb(esp_spp_cb_event_t event, esp_spp_cb_param_t *param)

              
variable or field 'my_spp_cb' declared void

bdring, 2018-12-05:
You need to get the latest Arduino Core

bdring, 2018-12-07:
Did this solve the problem? I would like to close this issue.

moromusa, 2018-12-07:
Updated and working!
Thanks a lot

closed 2018-12-07 by bdring