bdring/Grbl_Esp32 #67: LaserWeb 4 and Grbl_Esp32

KrisKelvin1, 2018-12-08:
how to make friends LaserWeb 4 and Grbl_Esp32 via WiFi ?

luc-github, 2018-12-08:
Use telnet is not working ? 

[image]

KrisKelvin1, 2018-12-08:
not working !

luc-github, 2018-12-08:
what's happen ? so details would help

KrisKelvin1, 2018-12-08:
I click "connect" and after a while comes out with a message

-----------------------------------------------
Connecting Machine @ Telnet,192.168.1.101
 Machine connected
 Work Offset: 0 / 0 / 0 / undefined
 No supported firmware detected. Closing connection to 192.168.1.101
 Machine disconnected
-----------------------------------------------

luc-github, 2018-12-08:
looks like only smoothieware is supported but here a guy succeed by emulating the Grbl 1.1f welcome string
https://github.com/LaserWeb/LaserWeb4/issues/454

Need to dig laser web code to see what welcome string is expected

luc-github, 2018-12-08:
Laserweb expect to see  `Grbl 1.1f ['$' for help] ` when connection start
But currently nothing is sent

KrisKelvin1, 2018-12-08:
And it turns out that it is impossible to work with the Laserweb program through the wifi?

luc-github, 2018-12-08:
Nope it is LaserWeb Limitation - lasergrbl  is working fine on telnet (https://github.com/bdring/Grbl_Esp32/wiki/Using-Telnet)

 I am checking the impact of adding adding this welcome string in telnet grbl_esp32

luc-github, 2018-12-08:
I have a solution but I need to clean the code :
on connection telnet send welcome string :
[image]

so now laserweb seems happy :

[image]

I need to discuss with @bdring  if we implement this or not

luc-github, 2018-12-08:
to clarify there are 2 ways to solve the issue 
1 -  as lasergrbl  do currently, laserweb should send Ctrl-X to reset grbl when connected and get this welcome string.
2 - or by default send the welcome string on connection, in that case better to  disable the ctrl-X in lasergrbl to avoid twice welcome message - but this is just cosmetic

luc-github, 2018-12-08:
Ok I can add a define to enable the telnet welcome string
By default it will be disabled

KrisKelvin1, 2018-12-09:
it will be very good if there is complete compatibility with LaserWeb4 , the program is very easy to use and I think that many users will be grateful to you!

luc-github, 2018-12-09:
@KrisKelvin1 The commit is in PR now :https://github.com/bdring/Grbl_Esp32/pull/62/commits/aac8f65ab6977ac5c3c3e1431884f50c88310dc4

I suggest you to open a ticket to LaserWeb4  as I am not sure this code will be integrated as @bdring wrote: send a ctrl-X is the method already used by other senders like lasergrbl, so this PR is a workaround just for LaserWeb and may not be kept for long time

openbuilds-engineer, 2019-01-02:
@cprezzi

luc-github, 2019-01-06:
@cprezzi yes I think would be great
Also may I know how do you control data flow on telnet in LaserWeb ?

DE-Koodie, 2019-01-06:
Im also planning to use an ESP32 with GRBL and Laserweb, Hardware is arriving this week - let me know if you need any help. Ported my Co2 to GRBL, i have an osciloscope and am familiar with basic programming.

cprezzi, 2019-01-08:
Ok, the actual lw.comm-server (4.0.133) has been extended to support the "RESET_ON_CONNECT" feature also for Telnet and ESP8266 connection types. It depends on @jorgerobles when he will compile the new binaries to update the installers.

cprezzi, 2019-01-08:
@luc-github The flow control is the same as with USBserial and differs slightly depending on the detected firmware. Basically we track the available space in the firmware buffer (by counting sent bytes and responses).

luc-github, 2019-01-08:
@cprezzi thank you for the feedback ^_^

bdring, 2019-01-08:
@cprezzi 

I think reducing the number of times data is sent by putting as much data as possible into each transmission could improve performance. Is there a way we could work together to achieve this? We can increase the RX buffer size.

luc-github, 2019-01-08:
@cprezzi I am guessing you use this calculation : https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface#streaming-protocol-character-counting-recommended-with-reservation

It is what LaserGRBL call buffered method

cprezzi, 2019-01-09:
@bdring A bigger RX buffer in the firmware could help with continous flow but I would need a way to detect that it's Grbl_Esp32 instead of normal grbl to adjust our internal counter. You could probably change/extend the welcome string slightly (like `Grbl 1.1f on ESP32 ['$' for help]`) so the frontend can detect it. What do you think?

bdring, 2019-01-09:
@cprezzi I would be concerned the welcome string change might confuse other senders. We could send a message string like this right after the welcome string.

[MSG:RX Buffer size,254]

cprezzi, 2019-01-10:
@bdring I don't know about other senders, but I guess most of them will just parse for a part of the welcome string like `Grbl 1.1f` (if the even check it). You could also add `;on ESP32` to the end of the string, because everything after `;` should anyways be treated as a comment only. 
Adding an additional string which is only used for Grbl_ESP32 and will slow down the detection process is no option for me.

cprezzi, 2019-02-19:
I got my ESP32 CNC Controller V3.1 and started testing. Unfortunately I only have ESP32 DevKit V1 with 30 pins, which do not match the pin mapping of the NodeMCU-32S. So I had to order some NodeMCU-32S :)

Nontheless I was able to programm the WebUI branch to one of my ESP32s and test it with LaserWeb4. After setting RESET_ON_CONNECT=1 in LW4s .env file, everything seems to work fine. Speed via USB and Telnet seems to be equal.

bdring, 2019-02-20:
@cprezzi That sounds great. 

@luc-github said there were some recent improvements to the ESP32 core that could improve telnet.

cprezzi, 2019-02-20:
I installed the "esp32 by Espressif Systems Version 1.0.1" board via the board manager (after adding the https://dl.espressif.com/dl/package_esp32_index.json url to the presets, which means I did not get the newest dev version. 
What I realized is, that when in AP mode, it's not possible to connect with my mobile (Android 8), but it works with a Win10 notebook.

bdring, 2019-02-20:
@cprezzi I would use the Git method to install the Espressif files. That allows you to easily get the updates. 

On Win 10, the web browser auto connects to the ESP32 when I connect to the AP. On my Android phone, I need to manually put the URL into the browser. The default URL in AP mode is 192.168.0.1

Geekoid85, 2019-07-14:
Hi there. 
I'm making some more testing on the firmware planning on using it to upgrade a charlyrobot cra4 cnc router from a makerspace. Really it's much more stable and reliable now, no problems what so ever to flash it and make some test with ugs (universal gcode sender). Only when I'm trying to use it with a laserweb usb connection it failed. Telnet is fine but I would like a wired connection to the pc.

What I don't understand is that in ugs I see the welcome message send by the esp. But in laserweb nothing so like said on top the connection is closed. Here is some screenshot, if someone succesfully get laserweb working in usb let me know !
[image]
[image]

cprezzi, 2019-07-15:
UGS probably sends a reset command after each connect. In LW4, you need to configure "RESET_ON_CONNECT=1" for that. We don't do that by default, because some machines don't like that.

bdring, 2019-07-15:
An Arduino reboots every time you connect. The ESP32 does not. If you want the reset message, the sender needs to request it as @cprezzi explains.

Geekoid85, 2019-07-15:
Okay that's a thing to know. Now that I've created the .env file with the right parameter as @cprezzi explained it work !
[image]
I encounter a similar problem in cncjs. All the jogging controls are disabled. After opening the usb connection like normal a click on the red reset button of cncjs fix it. I think technically it's the same problem. Grbl does not send the reset message so cncjs won't correctly connect. Maybe it would be good idea to reference this issue in the trouble section on the wiki.

Because it was not so easy on windows here is a guide :
[Documentation for the .env file](https://github.com/LaserWeb/lw.comm-server/wiki/Environment-File-(.env))
[image]
You want to create a .env file without giving it any name what so ever. 
On windows simply open Microsoft Notepad, copy past the example config of the documentation and modify the "RESET_ON_CONNECT=0" to "RESET_ON_CONNECT=1". Now you go to file, save as, select "all file (*.*) under the name box and write .env in the namebox.
Save it in the installation folder of laser web and you're good to go !

bdring, 2019-07-15:
I will add a note to the FAQ in the wiki.