bdring/Grbl_Esp32 #79: Status report won't switch to inches, and pin state question.

109JB, 2019-01-11:
New user of the Grbl ESP32 firmware.  First off, let me say that what I think you have done is fantastic. I have been using Grbl for several years on arduino and am excited about the ESP32 version as I have several times run into a wall with regard to step rate.

Anyway, I got a ESP32 dev board in the mail today and immediately used the Arduino IDE to upload the firmware. No problems with install. Here is the $I info:

[VER:1.1f.20180917:]
[OPT:VHSW]
[MSG:Mode=STA:SSID=:Status=Not connected:IP=0.0.0.0:MAC=B4-E6-2D-BF-B3-DD]
[MSG:Mode=AP:SSDI=GRBL_ESP:IP=192.168.0.1:MAC=B4-E6-2D-BF-B3-DE]

I have my own GUI that I use with the arduino version of Grbl and it connected fine and most things seem to work ok, but I do have a couple things.

1. When sending a $13=0 or $13=1 command to set mm or inch display units, it doesn't change and the units displayed are always in mm.

2. when looking at the status reports, I see that several of the input pins show activated. Namely YZRHS (Limit Y, Limix Z, hold, soft-reset, and cycle-start pins, ) Now being new to the ESP32, maybe it has something to do with running this with nothing hooked up to the ESP32. Mine doesn't even have header pins soldered on yet. Do you think maybe those pins are just floating and resulting in those indications? 

<Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:YZRHS|WCO:0.000,0.000,0.000>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:YZRHS|Ov:120,50,80>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:YZRHS>

3. Although I didn't have any problems with it, you may want to consider having the same default settings as the main Gbrl branch for the release download. I could see where this could confuse some people.

Finally, once again, what you have done is great.

109JB, 2019-01-11:
So I figured the I/O pins out, so #2 in my previous post is solved

#1 (status report doesn't switch to inches with $13) is still a problem. G20 and G21 commands still respond as expected, it is just the status display units that are not changing from mm to inches.

bdring, 2019-01-11:
I will look at the inch vs mm problem. I never use inches, so I probably never properly implemented that.

bdring, 2019-01-15:
Check the WebUI branch. I have updated to fix inch reporting.

closed 2019-01-15 by bdring

109JB, 2019-01-17:
Thanks for the fix.  It works, but in the "official" Grbl, inches have always reported 4 decimal places and mm 3 decimal places. 4 decimal places makes sense since 0.002mm ~= 0.0001inch 

I'm not that great at C-programming, but I was able to decipher this part anyway. I saw that your version hard codes the decimal precision to 3 decimal places, so I went ahead and changed report.cpp in my downloaded version. I just added an If-else , replacing line 117 of the WebUI branch report.cpp with this:

		
		if (bit_istrue(settings.flags,BITFLAG_REPORT_INCHES))
			sprintf(axisVal, "%4.4f", axis_value[idx] * unit_conv);  // Report inches to 4 decimals
		else
			sprintf(axisVal, "%4.3f", axis_value[idx] * unit_conv);  // Report mm to 3 decimals
		
I'm not sure if this is the best way to accomplish this, but I do think it should be added for consistency and practicality for those using inches.