bdring/Grbl_Esp32 #84: How to invert select control pins

109JB, 2019-02-02:
So I'm about ready to switch out my AVR Grbl for the ESP32 Grbl and put it into use on a CNC converted milling machine I have.  I have one thing I would like to do but due to my lack of C programming knowledge I don't know how to do it. What I want to do is invert the Safety Door input pin but not the other control pins.  In the AVR Grbl the Config.h file had the following which would allow this:

// Inverts pin logic of the control command pins based on a mask. This essentially means you can use
// normally-closed switches on the specified pins, rather than the default normally-open switches.
// NOTE: The top option will mask and invert all control pins. The bottom option is an example of
// inverting only two control pins, the safety door and reset. See cpu_map.h for other bit definitions.
// #define INVERT_CONTROL_PIN_MASK CONTROL_MASK // Default disabled. Uncomment to disable.
// #define INVERT_CONTROL_PIN_MASK ((1<<CONTROL_SAFETY_DOOR_BIT)|(1<<CONTROL_RESET_BIT)) // Default disabled.

I see that in the ESP32 Grbl, this has been changed, and I saw the following note in cpu_map.h

Unlike the AVR version certain pins are not forced into the same port. 
Therefore, bit masks are not use the same way and typically should not be 
changed. They are just preserved right now to make it easy to stay in sync
with AVR grbl

So, is there currently a way to invert only select control pins while not inverting the others?

bdring, 2019-02-02:
The control pin inputs can be any IO pin. When they are read, they combined into a status byte. system.h defines the bits...
```
#define CONTROL_PIN_INDEX_SAFETY_DOOR   bit(0)
#define CONTROL_PIN_INDEX_RESET         bit(1)
#define CONTROL_PIN_INDEX_FEED_HOLD     bit(2)
#define CONTROL_PIN_INDEX_CYCLE_START   bit(3)
```
system.cpp has this in the function that reads the pins
```
#ifdef INVERT_CONTROL_PIN_MASK
    control_state ^= INVERT_CONTROL_PIN_MASK;
#endif  
```

in config.h you can define the mask like this...
```
#define INVERT_CONTROL_PIN_MASK  (1<<CONTROL_PIN_INDEX_SAFETY_DOOR) 
```
.. or'ing together more pins with "|" 

I have not heard feedback from people doing this, so hopefully it all works OK.

BTW: When I have been configuring machines lately, I have been putting a lot of defines in cpu_map.h. That makes it easier to keep as many changes in one spot as possible.

109JB, 2019-02-03:
Thanks. I'll give it a try  and let you know how it goes. I tried testing it but I think I have some kind of IDE issue because after changing the one line on config.h, it is telling me that the program excess the esp32's memory capacity. Once I get that sorted I'll be able to test the control pin mask.

109JB, 2019-02-03:
Well I guess I did something wrong with the pull request because the second file modified didn't come through. I had also deleted the mask from cpu_map.h and it didn't show up on the pull request.  I had thought the pull request would compare all files in my fork to yours and include all changes to the pull request.  Do you know what I did wrong?

bdring, 2019-02-03:
Thanks for all the work on this. 

I think originally I did not support the INVERT_CONTROL_PIN_MASK and then did not clean up the notes.

I'll take care of getting the changes in. I don't know what went wrong for you.

closed 2019-09-10 by bdring