bdring/Grbl_Esp32 #86: Serial on pins?

109JB, 2019-02-04:
Maybe just a question.  I am using a NodeMCU ESP32S board and I see TX0 on GPIO1 and RX0 on GPIO3.  Are these pins connected to the USB to serial converter like the Arduino's

The reason I ask is because in the past I had 2 arduinos connected TX-RX & RX-TX with one running Grbl and one running with a touch display for a headless interface for the GRBL AVR. I would like to try the same thing with the ESP32 Grbl.

bdring, 2019-02-04:
Those are used for USB UART, but you can create another TTL UART with just about any free pins.

109JB, 2019-02-04:
So I should be able to upload and configure the ESP32 using USB and then connect the GPIO pins 1&3 to the other MCU and use them for serial com between the two as long as I am not using the USB at the same time.  I my understanding correct?

bdring, 2019-02-04:
No.

The ESP32 has 3 serial ports. Leave the current one connected to the USB alone.

Define a new UART on different IO pins and use that for your apication.

Google "ESP32 Arduino Second Serial Port" for help

109JB, 2019-02-04:
I think you misunderstand. When using the pins for external comm, the USB wouldn't even be connected.  Only external power to the Vin pin. There wouldn't even be a USB cable plugged in. I would only use the USB cable for flashing and when doing that would unplug the other MCU from the GPIO 1&3.

109JB, 2019-02-04:
OK.  Thanks for the explanation. I'll look into adding an additional port.

closed 2019-02-04 by 109JB

Protoneer, 2021-05-20:
@bdring would it be possible to have two pins (13,14) being able to do UART or I2C? (Based on machine file)

bdring, 2021-05-20:
It requires extra programming to define what those pins do.

The OLED is an example. See the Custom folder.

Protoneer, 2021-05-20:
@bdring Thanks!!!