bdring/Grbl_Esp32 #87: Breakout circuit for driving 5V stepper drivers

109JB, 2019-02-04:
Not an issue, but I wanted to pass along the circuit and picture of the board I made for driving 5V inputs on larger stepper drivers. Some of the drivers I use have optocoupler circuits built into them that I have found sometimes don't work well from 3.3V MCU outputs. So I used some stuff I had on hand to make a breakout board for converting the logic levels. 

I first tried using some of the cheap logic level shifters that are sold everywhere and while they worked, I saw significant rounding of the leading edge of the signals on my hand-me-down oscilloscope. At higher frequencies the signal essentially became a sawtooth rather than a square wave. I was concerned that this might cause problems if used on the step and direction pins, so I dug in a box of stuff I got from my father-in-law and found some 74HC245 transceivers. These turned out to work well for one-way logic level shifting of the outputs. 

For the inputs to the MCU I didn't have the concerns about clipping of the leading edge, so I used the cheapo logic level shifters because their circuit also includes a pullup resistor. Since the GPIO 34-39 pins require an external pullup anyway it seemed like a good choice.  I came up with the following circuit and put it together on a prototype board and it is working very well. I thought that perhaps others might need something like this.  

This circuit provides for 16 outputs and 8 inputs, I did my pin mapping to account for future expansion to 4 axes with spindle PWM, spindle direction, mist coolant and flood coolant enabled. The 74HC245's are wired for one-way communication so are only good for outputs. Since the clipping concern is only in regard to the step and dir  signals I probably should have used 2 of the level shifters and just one 74HC245. This would have allowed having 8 pins dedicated to step and dir (4 axes) and 16 pins that could be configured as either inputs or outputs.  Oh well, the circuit as shown works for my intended application which is a CNC milling machine. I also pulled out the EN pin and the GPIO1 & 3 to a separate header so I can wire in a headless controller and an external hard reset button.

Bench testing on the oscilloscope and with real hardware shows nice square waves on the outputs and the inputs all work as expected so I'm happy with it.  Anyway, here is the wiring and a picture of the prototype board. I used a NodeMCU ESP32S MCU with the pinout as shown and disabling all of the internal pull-ups in the Grbl ESP32 firmware. I was thinking about having a PCB made, but the layout is pretty clean on the proto board and I only need one of these. I may try milling a PCB later, but for now this will work great. One thing I'm going to add is a 5V regulator circuit in the lower left for external board power as now everything is powered from the USB. Anyway I hope this can be helpful to someone else.

[image]

[image]

bdring, 2019-02-04:
Thanks for the input.

In my experience , the 3.3v is not enough to turn on the optos. The stepper drivers have an internal resistor scaled for 5V.

A lot of basic level shifters cannot supply the current to turn on the optos. They often use a pull up resistor for the logic high and that won't work for the optos.

I simply use an NPN to switch the low side of the opto to ground and directly hook the high side to 5v. It seems to work perfectly.

cyberhasse, 2019-11-29:
Hi!
Just bought 2 of your board and finished connecting my TB6600 drivers as i had them connected,
to my arduino nano..
Motors are not moving and i suspect it is all about 5v vs 3.3v..
how would one go about fixing this problem?
Great boards btw. and awesome Grbl version :)
Regards.