bdring/Grbl_Esp32 #90: 4/5 axis questions

109JB, 2019-02-08:
So I've managed to modify the WebUI branch code to include 2 additional axes, A and B and they are working, but there is something I would like to attempt to do.  I am not very good at C/C++ programming and would like a little assistance on where to start looking in the code to accomplish this. I just want a tip or two on where to start looking and won't barrage with countless questions. Here is the issue I want to attempt to correct:  

The "norm" for rotational axes is that if there is a linear (xyz) axis on the same line as a rotational move (ab) that the feed rate should be calculated based only on the linear axes with the rotational axes following in a coordinated fashion to complete when the linear axes complete. If there is no linear axis in the line of code the motion should be calculated based on the rotational axes. For example, a G91 G1 X1 A1 F1 command should complete in one minute and a G91 G1 A1 F1 should also complete in one minute. The way it is currently working, the G91 G1 X1 A1 F1 results in a move that takes 1.414 minutes because the A axis is included in the calculations.  In conjunction with this, feedrate reporting should coincide with the above where the reported feedrate is based on only XYZ if present or on AB if only rotational. Where would I start looking in the code to see about "correcting" this?

bdring, 2019-02-08:
I would suggest asking this question at the Gnea Grbl Github repo. Sonny might have some good input.

I have used Grbl 3+ axes before for robotics and I wanted the pure linear feed rate.

109JB, 2019-02-08:
OK.  I can see how for robotics that the linear feed rate would be preferred.  The other option I have is to do this at the GUI level.  I'll see what Sonny has to say.

closed 2019-02-09 by 109JB

aldanstar, 2019-08-12:
I saw that 4 axis high current CNC controller is builded by you about month ago. When PCB will be officially released with updated firmware? Is firmware will be uploaded to this repo or new?