bdring/Grbl_Esp32 #92: Unexpected error in gcode file

ggallant571, 2019-02-10:
The attached grbl test file generates error messages using esp32 grbl but not with Arduino. Both are same version. I am not sure if it is the controller or host system but the "OK" responses seem to arrive in bunches. The error text is appearing twice with characters intermixed. The exact code is usually jumbled but 2 and 24 appear to be good candidates.

The file is intended to burn 4 patterns on low power laser. It uses the grbl arc commands. My test is with just 2 stepper motors. I isolated each of the 4 shapes and individually they complete without error.

For the short term I am:
    1. Looking to enlarge the input buffer.
    2. Change host TX scheme such that it waits for the "OK" before sending a new command
    3. Test on a Windows 10 machine

[test.zip](https://github.com/bdring/Grbl_Esp32/files/2849142/test.zip)

109JB, 2019-02-10:
I can run your code on my esp32 using my own GUI with no errors.

My gui uses the send-response protocol where it sends a line and then waits for "ok" before sending the next.

Are you sending a new-line, carriage-return or both when sending data?  I only send the new-line.  Also you have to count these when implementing character counting.

If the responses are garbled like you say then there is either something wrong in your code that is reading the serial data. In any case it appears it is GUI related.

ggallant571, 2019-02-11:
Thanks for testing. I will try again in single message per ok mode.

ggallant571, 2019-02-11:
That file had 0x0D 0x0A for eol. Changed to just 0x0A and it works. Easy fix on my end.

ggallant571, 2019-02-11:
Problem was 100% my host application. Firmware was ok.

closed 2019-02-11 by ggallant571