bdring/Grbl_Esp32 #96: Spindle direction pin

bovo-fabiano, 2019-02-20:
Hello all!
I have an Arduino Grbl project in which I use Coolant, Mist and Spindle Enable, setting those in the config.h as usual. Flood stills in pin A3, Mist goes to pin A4 and Spindle Enable goes to pin D13 leaving PWM on pin D11.

I noticed Grbl_Esp32 has the same options but I wonder what pin to use as Spindle Enable.

In cpu_map.h line 59 it shows Coolant_Flood on GPIO_num_**16** and Coolant_Mist (default commented) on GPIO_num_21. In lines 83 and 84 it shows Spindle_Enable and Spindle_Dir (both default commented) on the same GPIO_num_**16**.

I don't need spindle direction but my project needs a Spindle Enable.

How could I set that?

Thanks in advance!

bdring, 2019-02-20:
Is this a custom design or are you using an existing pcb?

You can use any pin that is available and can be an output. Are you using the probe feature? You could use that pin. Maybe GPIO_NUM_21 or 21?

Did you see this page?

https://github.com/bdring/Grbl_Esp32/wiki/Setting-Up-the-I-O-Pins

bovo-fabiano, 2019-02-20:
Thanks for your prompt reply Mr. Bart!

I have a custom PCB much like yours for the Arduino Nano I use. It's a multi purpose XYZ machine for personal use. I don't have a inside picture but you can see the machine here: https://youtu.be/cgXZaWFyrTY

What about using:

GPIO_NUM_16 for Spindle Enable
GPIO_NUM_17 for Spindle PWM
GPIO_NUM_21 for Coolant Mist
GPIO_NUM_22 for Coolant Flood
GPIO_NUM_32 for the Probe

I see that using #21 and #22 would prevent me to use LCD in the future but it is not a problem for me as I control it with an old computer running Linux. In this same thought I wouldn't need the SD card and ther would be many pins available.

Thanks!

bovo-fabiano, 2019-02-20:
Ok! I will keep the SD Card at first.

Considering your board, does it work with that ping configuration?

Have you wondered adding relays to the M3, M7 and M8 commands? That's what I did in my Arduino Board.

Thanks,
Fabiano Bovo - Brazil

bdring, 2019-02-20:
GPIO 20 and 21 don't have connector pins on my board. Other than that you pinout is OK

My board is intended as a general purpose development board. I don't want to make it too specific for one application.

closed 2020-01-10 by bdring