bdring/Grbl_Esp32 #100: Virtual serial port

HellKern, 2019-03-06:
Hey,
Is there option to use Grbl_Esp32 like vanilla GRBL - using PC software, that sends G-Code to the machine, but using some serial-over-IP functionality. I understand that it will add 30-100ms delay for each line in G-Code, but i think that this option would be nice to be.

luc-github, 2019-03-06:
Sorry what is the difference with telnet feature ?https://github.com/bdring/Grbl_Esp32/wiki/Using-Telnet

HellKern, 2019-03-06:
Difference is in compability with software, that can use only serial port to communicate with machine, currently i am in searh of solution, that will create virtual serial port on PC and will translate commands to this port via telnet, but actually i will not be able to test it untill assembe esp32-driven CNC controller

luc-github, 2019-03-06:
So this is not related to ESP32 neither grbl_esp32 projects but more about 3rd parties solutions doing serial to socket like socathttps://unix.stackexchange.com/questions/453974/create-a-udp-to-serial-bridge-with-socat or tcpcom https://sourceforge.net/projects/combytcp/
Have several solution available but it is out of grbl_esp32 scope IMHO

closed 2019-03-20 by bdring