bdring/Grbl_Esp32 #99: Spindle/laser connection

HellKern, 2019-03-06:
Hey,
This question is not about the software, but i think this is the correct place to ask.
I am going to prototype own board to use Grbl_Esp32 and faced with issue driving spindle/laser from ESP32 - 3.3v-50mA output from EPS32 pins is to low to drive powerful mosfets, so maybe someone already have tested solution how to drive 10-100watt load from ESP

easytarget, 2019-03-06:
I'd start by considering a different mosfet.. logic level ones come in a wide range of power capabilities, eg if you want 100W from a 12v load an [IRL530](http://www.irf.com/product-info/datasheets/data/irl530n.pdf) will do that effortlessly (with a heatsink..)
So long as it's a logic-level mosfet the 3.2v from an esp will turn it fully on, this is what they are designed for.

If you need to use a different mosfet, then add an optoisolator (4n35 or similar) to both isolate and amplify the signal to the full voltage of the drive; or a simple transistor amp will do. Plenty of instructions and circuits on the web for this; try: https://www.google.nl/search?q=irl530n+opto+isolator+arduino&newwindow=1&tbm=isch

HellKern, 2019-03-06:
> I'd start by considering a different mosfet.. logic level ones come in a wide range of power capabilities, eg if you want 100W from a 12v load an [...]
It's quite tricky about that, because ATmega and ESP output rates are different and people facing with issues when trying to drive mosfets directry from ESP, for example - internal capacity of mosfet is big enough to burn out output cascade of ESP after some time of work. Thats why i am asking if anyone have a tested solution...

bdring, 2019-03-06:
What are you powering with the MOSFET? I have been using a IRLR120N on a design for a while to drive a solenoid.  So far so good. I use it with a 5kHz PWM.

easytarget, 2019-03-06:
> It's quite tricky about that, because ATmega and ESP output rates are different.

Basically, some controllers are 5v, and some, like the ESP module, are 3,2v. There are lots of 3.2v logic microcontrollers about, Look at circuits for driving powerful loads from the Raspberry PI for instance. And get some background on mosfet logic switching from http://www.vishay.com/docs/49940/49940.pdf

For some tested examples and practical info check out how 32bit 3d printer controllers go about switching their heated printbeds; 20A systems are quite common..  The 'Gen 7 reprap board is 3.2v; and there is a bunch of info on this here: https://reprap.org/wiki/Gen7_Research#MOSFET_heat_and_PWM

Please tell us what you are driving (voltage? is it a Motor/Solenoid/Heater? and are you aiming for a specific pwm frequency) since, as you can see from the above links, I do not think anybody can give clear help without more info from your side.

HellKern, 2019-03-08:
@easytarget great idea to take a look at 3d controllers chematics, thanks a lot!
Currently i am going to drive 36V/2A spindle and 12V/2.5A laser module, but i am trying to do some universal solution(current arduino-based solution drives both spindle and laser using one IRF540 mosfet - "ground" wire of both devices are going throw mosfet, and power wires are going from different powers sources)

bdring, 2019-03-08:
Typically lasers and variable speed spindles are controlled with a TTL PWM signal. For On/Off control, I suggest a relay.

HellKern, 2019-03-09:
Well, i asked, i answering - for one's with same issue. Hope it will help - there are MOSFET drivers existing, and it's very easy to drive almost any MOSFET with them, i've tested MC34152 and IR4427 on a bradboard and ESP32 sucsesfully drive all MOSFETs i had neadby with very good signal shape at load up to 10kHz PWM, and PWM still was working up to 100kHz, but signal shape wasn't square anymore.
@bdring typically CNC machines are driven with industrial grade controllers, we are talking about on-the-knees builds here ;)

closed 2019-03-09 by HellKern