bdring/Grbl_Esp32 #101: Include $N functions

GandalfOp, 2019-03-07:
I'm interested on using the $N Startup function from GRBL using your board and your code (must say it works wonderful to be honest!!)

I've seen that the "original" GRBL, can use the $N Functions to assign a "startup" functions such as Homing, or Assign mm/inch etc...

I think it would be useful to use that function to send some code in terms of "self-working" machines, that run over and over the same code.

(I plan to use it personally to send the "$FM, $F, and $F/heremygcode.gcode" codes needed to start a gcode stored in the microSD)

Thanks!

bdring, 2019-03-07:
I will look at $N feature soon. I am going on vacation for almost 2 week, so it could take a while.

The $FM style SD card functions are being phased out in favor of {ESP...] style functions. This was to deal with Grbl's character restrictions and other reasons.

GandalfOp, 2019-03-07:
Of course!

Have a good time meanwhile!!

GandalfOp, 2019-03-28:
Dear bdring, hope your vacation was great! 
I've kept on trying to improve my autonomous machine, but suddenly, after a while I found out that sending my gcode (streaming it) from UniversalGcodeSender works well, but saving it to the SD, then using $FM, $F, and $F=/mygcode.gcode sends out an error on this last code. (error:1 in SD file at line 1) attaching it here just in case. Does the UniversalGcodeSender "modify" it before sending it to the ESP32?

[Shamrock2.txt](https://github.com/bdring/Grbl_Esp32/files/3019739/Shamrock2.txt)
If that's the point, how should I code my gcode for it to be undestood by the ESP32?

On your last comment, you say that the $FM style SD card functions are being phased out in favor of {ESP...] style functions. does that mean that my 3 codes ($FM, $F, $F=/mygcode.gcode) will change to something like ESP_mount, ESP_play?

Other question, unrelated to the topic, (if you think it needs a specific issue feel free to delete it from here).
I undestand (from my poor knowledge) that the $FM for example, is related to a memory adress (i.e 0x00, 0x01, 0x01, 0x10, etc... ) if that's the case, where can I find that 3 addresses to execute them from the code (using an interruption for example)

Thanks a lot! and appreciate your work so much!

bdring, 2020-01-10:
On roadmap...closing

closed 2020-01-10 by bdring