bdring/Grbl_Esp32 #103: Cannot get it to install at all

Slixxor, 2019-03-11:
I am using a NodeMCU V3, I have the master ZIP, It wont install directly into Arduino IDE that's Ok, I've found the project file. It runs OK. When trying to upload it says "driver/rmt.h: No such file or directory"

I have searched the net and found an updated rmt.h file and manually created one in the project folder. This doesnt work. I've created a folder called "driver" then added it there, that also doesnt work. 

i then commented the line out and hit another error immediately after. "esp_task_wdt.h: no such file or directory"

Ok let's comment that out and see how deep this rabbit hole of missing files goes..

Turns out they are all pretty much missing. I'm a bit confused here. How can I get this working as expected?

bdring, 2019-03-11:
I sounds like you have not I stalked the ESP32 files.

https://github.com/bdring/Grbl_Esp32/wiki/Compiling-the-firmware

Slixxor, 2019-03-11:
right. well i've followed all the instructions again and same issue. Stalls at line. #include <driver/rmt.h> saying no file or directory.

IDE version 1.8.8

Just to confirm, 
- I've added http://arduino.esp8266.com/stable/package_esp8266com_index.json to my additional boards in the IDE.

- I've installed  the board package from board manager and my  ESP32 board options are appearing.

- Ive downloaded arduino-esp-master.zip and tried installing it via ZIp import in library manager. that failed so i manually copied all folders to my documents./Arduino/libraries folder and overwrote every single folder to make sure.

- I've copied the ESP32SSDP and arduinoWebSockets folders to my Documents/Arduino/libraries folder

- Set my board to NodeMCU 1.0 and tried uploading a blank sketch.. SUCCESSFUL

- I've downloaded Grbl_Esp32-master.zip and extracted the files.

- I've launched the Grbl_Esp32.ino in the libraries folder that was in the Grbl_Esp32-master.zip 

- Set my board to NodeMCU 1.0 and tried uploading,...FAILED at driver/rmt.h: No such file or directory

-  Tried uploading a blank sketch.. SUCCESSFUL

Am I Missing something?

Slixxor, 2019-03-11:
Whoops. i see the error there. Wrong ESp library in boards manager. I cleared the wrong one. Now resolved. New bug..

esptool.py v2.6-beta1
Serial port COM4
Connecting....

A fatal error occurred: Timed out waiting for packet header
A fatal error occurred: Timed out waiting for packet header

I've tried selecting NodeMCU-32S with both 921600 and 115200 baud rates same error.

Slixxor, 2019-03-11:
Please see:

The board i'm using front: https://libraries.sharpsoft.com.au/dev/board.jpg
The back of the board: https://libraries.sharpsoft.com.au/dev/board_back.jpg
Settings: https://libraries.sharpsoft.com.au/dev/settings.jpg
Output from build: https://libraries.sharpsoft.com.au/dev/output.txt

luc-github, 2019-03-11:
This an esp8266 board it is wrote on it. It is not an esp32 board so you cannot flash ESP32 FW on ESP8266 board

Slixxor, 2019-03-11:
> 
> 
> This an esp8266 board it is wrote on it. It is not an esp32 board so you cannot flash ESP32 FW on ESP8266 board

OK..

Is it possible to get it to work on a NodeMCU? This is an ESP8266 but it is a NodeMCU. Any GRBL flavours for this?

luc-github, 2019-03-11:
Sorry not as I am aware of. ESP8266 is not powered enough and not enough pins available neither

Slixxor, 2019-03-11:
ok. I've ordered some ESP32S boards instead :)

closed 2019-03-11 by Slixxor