bdring/Grbl_Esp32 #105: Doubts about functionalities

XngelGT, 2019-03-19:
Hello,

I'd like to use GRBL for comunication with the Universal GCode sender program in Windows. I've managed to establish connection with it, so far so good. However, for the application I'm developping, I need to only get the GCodes from the program into the ESP32 (e.g to receive into the ESP32 "G00 X3.2 Y4.5 Z5"), and not the steps for driving motors. Could you give me some tips or insight on how could I achieve this?

Thanks for your answers!

ggallant571, 2019-03-19:
I am quite confused by your question. The "G00 X3.2 Y4.5 Z5" text is GCODE that requests  GRBL to drive the 3 motors.

XngelGT, 2019-03-19:
I understand that I'm not being as explicit as possible. I'll try it again.

- I have a GCode for manufacturing something

- I use Universal GCode Sender to connect via GRBL with my ESP32

- I use UGS to send each line of the GCode to my ESP32 and monitor my CNC machine
(Here's where it gets tricky)

I want to receive the line UGS sent in my ESP32 without the interpreter turning it into steps for my motors. Meaning I need to get in the ESP32 the GCode line by line. 

I hope this clarifies my doubt, if not I'll keep trying :D

bdring, 2019-03-20:
I am not sure what you are trying to do. Please tell us why and maybe we will understand.

As the characters arrive one at a time, Grbl parses them to remove unused characters and immediately act upon real time command characters. It puts the everything else into a buffer. Once a line end is detected in the buffer, it parses that line to see if it is legal and then executes it when it can.

You may be able to change that to do what you want. It is unlikely we would want to change any of the branches.

Most of the stuff is in serial.cpp and protocol.cpp

https://github.com/bdring/Grbl_Esp32/blob/WebUI/Grbl_Esp32/protocol.cpp#L110

cprezzi, 2019-03-21:
@XngelGT Grbl is a GCode interpreter/parser, what you seem to want is a serial port logger. But what you didn't say is what you want to do with the gcode lines. Do you want to write them to the SD card or parse some of them and to something?

XngelGT, 2019-03-21:
> @XngelGT Grbl is a GCode interpreter/parser, what you seem to want is a serial port logger. But what you didn't say is what you want to do with the gcode lines. Do you want to write them to the SD card or parse some of them and to something?

That is what I want to do, I already manage to do it with a bit of thinkering in the protocol.cpp function. Everything is so well organized that it makes it so easy. Now I want to know how to pause/resume execution of the job from the ESP32, is this possible?

XngelGT, 2019-03-21:
> That is what I want to do, I already manage to do it with a bit of thinkering in the protocol.cpp function. Everything is so well organized that it makes it so easy. Now I want to know how to [...]

Maybe it's important to mention my situation. I am in charge of a part of a larger project to develop a CNC machine. My part only includes the interfacing with a software as well as sending the GCodes via serial to another MCU which parses themm does the interpretation and the motor control. This is why I only wanted to get the GCodes, but now I need a way to pause the execution of the GRBL code so that I wait for the external MCU to finish with the previous task and then resume it to continue the manufacturing. 

I hope I'm not being too annoying and I deeply thank you for your answers.

bdring, 2019-03-21:
It seems very strange to be using Grbl for this. Isn't it better to write/modify a sender? 

In the most basic form, the senders are waiting for the "OK" from Grbl. You can control the flow that way. Some commands, like feed hold and feed rate override are "real time" and do not need an OK.

Your project sounds very outside the scope of Grbl, so I am going to step out of this conversation.

Good Luck

XngelGT, 2019-03-21:
So there's no way of pausing the execution of the GRBL? The thing is that we want to use brushless motors for our machine, hence the controller included in GRBL is not enough for us. I'll keep looking through the code, any help is useful.

Thanks!

closed 2019-03-25 by bdring

easytarget, 2019-03-25:
You may also want to check out:
https://awesome.tech/grbl-demystified/
- it will give you a better feel for grbl's architecture and why, as Boromir would say "One does not simply 'pause' GRBL"

cprezzi, 2019-03-25:
@easytarget Your post sounds like spam for your product.

easytarget, 2019-03-25:
Not my blog post.. nothing to do with me; I found it with ordinary Google-fu a couple of weeks ago when I was also wondering about how easy it would be to 'just' extend Grbl for something. And while it mentions their Grbl fork, it's hardly spammy about it, you need to go to their homepage for that.

Anyway, as you said; what the OP is trying to do sounds like simple text parsing and Grbl wouldn't be an asset ;-). But  in the spirit of being open and co-operative I thought it may help the OP or anybody else who finds this thread in the future with similar mad plans; it is of no use to those who already know the architecture, but coming to it cold I was really glad someone has taken the time to write this up.