bdring/Grbl_Esp32 #106: Setting wifi credentials

Slixxor, 2019-03-22:
Have you added dynamic wifi SSID and password entry for connecting to an existing network or setting up the AP over USB/Bluetooth? 

I'm assuming the commands from USB/bluetooth/wifi are parsed in the same location?

i.e: Add a command like $W=[SSID]/[Pass]/[AP bool]/[*DHCP bool]/[*ip]/[*subnet]/[*defgateway]

Where * can be null for nonAP wifi access and relevant settings.

I've suggested "/" as a delimiter because it can't be used for an SSID.

I've considered making the implementation myself but was wondering if you got around to it or something similar.

luc-github, 2019-03-22:
the [ESPXXX] command are listed here https://github.com/bdring/Grbl_Esp32/blob/WebUI/doc/Commands.txt

Slixxor, 2019-03-22:
Are these accessible via USB connection?

Slixxor, 2019-03-22:
Sharp Software is proudly an Australian owned and operated business dedicated to providing reliable software and electronic solutions with simple layouts that are easy to use, powerful and designed to improve our clients business.

Using the latest technologies, our experienced design team strive to bridge the gaps between People and Process saving you time and money.

Team Philosophy: Perseverance over time equals results.

Slixxor, 2019-03-23:
> 
> 
> the [ESPXXX] commands are listed here https://github.com/bdring/Grbl_Esp32/blob/WebUI/doc/Commands.txt

Are these accessible via USB connection?

Maybe my usage is incorrect, if i'm in the Arduino IDE serial monitor and I send [ESP800] I get an error code of 1.

Also, what is the default password for the commands?

bdring, 2019-03-23:
They will work through the serial port in the IDE.

The default password is "admin"

Slixxor, 2019-03-23:
yeah i'm having dramas. are you sure the implemetation os done can you confirm exactly what I would need to send to get anything to come up. Choose any command, am I  including '[' and ']' on each command or not? If I do i get error:1 if i don't I get error:2

Slixxor, 2019-03-23:
> 
> 
> They will work through the serial port in the IDE.
> 
> The default password is "admin"

[ESP105]<SSID>pwd=<admin password>

bdring, 2019-03-23:
This section of config.h controls the authentication and password.

> #define ENABLE_AUTHENTICATION
> 
> #define NAMESPACE "GRBL"
> #define ESP_RADIO_MODE "RADIO_MODE"
> 
> #ifdef ENABLE_AUTHENTICATION
> #define DEFAULT_ADMIN_PWD "admin"
> #define DEFAULT_USER_PWD  "user";
> #define DEFAULT_ADMIN_LOGIN  "admin"
> #define DEFAULT_USER_LOGIN "user"
> #define ADMIN_PWD_ENTRY "ADMIN_PWD"
> #define USER_PWD_ENTRY "USER_PWD"
> #define AUTH_ENTRY_NB 20
> #define MAX_LOCAL_PASSWORD_LENGTH   16
> #define MIN_LOCAL_PASSWORD_LENGTH   1
> #endif

With settings above, here are some examples.

> TX: [ESP800]pwd=admin
> RX: FW version:1.1f # FW target:grbl-embedded  # FW HW:Direct SD  # primary sd:/sd # secondary sd:none # authentication:yes # webcommunication: Sync: 81# hostname:grblesp
> 
> TX: [ESP111]
> Rx: 192.168.1.11
> 
> Tx: [ESP121]pwd=admin
> Rx: 80
> 
> Tx: [ESP121]pwd=admin
> Rx: 80
> 
> Tx: [ESP121]81 pwd=admin
> Rx: ok
> 
> Tx: [ESP121]pwd=admin
> Rx: 81

I am not sure where you got "testpwd". If you have some ideas to clarify the commands.txt. please send them and we will update.

Slixxor, 2019-03-23:
The command for ESP105 is stated as :
*  Set/Get AP SSID 
[ESP105]{SSID}pwd={admin password}

I was trying to set the SSID as 'test'

Also, i'm not sure whats going on. I still get error:1 the entire INO does not have any reference to the above code you supplied. 

[libraries.sharpsoft.com.au/sharpcnc/dev/wifi.jpg](url)

bdring, 2019-03-23:
The code is not in the .ino file. It is in the config.h file.

You should send....

[ESP105]test pwd=admin

You need a space between the parameters

bdring, 2019-03-23:
Are you using the WebUI branch of the code?

Slixxor, 2019-03-23:
? I'm not sure. Do you have a link?

Slixxor, 2019-03-23:
Sorry for hassling you, You've put so much hard work in, now you're helping plebs get it working :P

Slixxor, 2019-03-23:
https://libraries.sharpsoft.com.au/sharpcnc/dev/libraries.jpg

https://libraries.sharpsoft.com.au/sharpcnc/dev/sketch.jpg

Slixxor, 2019-03-23:
Found the link. Sorry I hate the way github is layed out.

Slixxor, 2019-03-23:
Ok, webui doesnt compile or install

please see error: https://libraries.sharpsoft.com.au/sharpcnc/dev/errorwebui.jpg

bdring, 2019-03-23:
[See the Wiki. Compiling The Firmware](https://github.com/bdring/Grbl_Esp32/wiki/Compiling-the-firmware)

Slixxor, 2019-03-23:
Just to confirm, Is it mandatory to have an SD card attached to the chip to get everything to work?

Slixxor, 2019-03-23:
I have followed the instructions. No go. I think I might can this project. I was planning on adding this to our custom controller board for speed, performance and functionality on our CNC mills we sell but this is proving to be rather challenging.

I'll pass it onto one of my engineers to try get it working or to bridge your master with the WebUI branch to make it easier to use. We have no need for the webUI features as we have our own GRBL compatible control software i've written. Just being able to setup bluetooth and wifi was all that we needed,

bdring, 2019-03-23:
An SD card is not required. We use a slack channel for real time chat. Here a an invite link.

https://join.slack.com/t/buildlog/shared_invite/enQtNTg3NDQ3MjQwMjE0LTVhOTVmNDRiNjU5OTFlNjdmMmZiNmExZDMwOWRjZWI5NTNhYzAzMWFjNmJmNTk3NmMwZGNhMjFmYjYyMzE3NGI

I don't know why you are having trouble. Most people have no issues if they follow the instructions. It is important to make sure you have the latest Arduino, and Espressif files.

I don't have time now for a long help session, maybe later next week.

Slixxor, 2019-03-23:
ok great. when you get a chance.

closed 2019-03-23 by Slixxor