bdring/Grbl_Esp32 #107: X-LIMIT floats instead of being pulled high

DavidBoman, 2019-03-22:
I'm trying to use limit switches to home my machine. I've connected normally open (NO) switches on X, Y and Z and connected them to the appropriate pins. The homing works for Z and Y but fails for X. grbl correcly show a change when the Y and switches are toggled but for X the value is always triggered. 

When measuring ON the pins connected to 100 Ohm resistance I can see that Y and Z are being pulled HIGH at 3.3V but X is floating at 0.2V. Electrically pulling the pin HIGH also toggles the status in grbl as expected. 

When reading the code it seems like the pin should be pulled high...
```cpp	
void limits_init()
{  

  #ifndef DISABLE_LIMIT_PIN_PULL_UP  
		#ifdef X_LIMIT_PIN
			pinMode(X_LIMIT_PIN, INPUT_PULLUP);  // input with pullup
		#endif
		#ifdef Y_LIMIT_PIN
			pinMode(Y_LIMIT_PIN, INPUT_PULLUP);
		#endif
		#ifdef Z_LIMIT_PIN
			pinMode(Z_LIMIT_PIN, INPUT_PULLUP);
		#endif
	#else
```
... but still it floats. Any ideas?

bdring, 2019-03-22:
What pin number are you using?

DavidBoman, 2019-03-22:
I'm using the board I bought from you on Tindie so it's the top left-most pins show on the picture on this projects main page (https://github.com/bdring/Grbl_Esp32). i.e. X-Limit-GND, Y-LIMIT-GND and Z-LIMIT-GND.

DavidBoman, 2019-03-22:
According to schematics on buildlog (http://www.buildlog.net/blog/wp-content/uploads/2018/10/schm_esp32_cnc_test_v3p1.pdf) that cooresponds to:
Y_LIMIT: GPIO04 (pin 26)
X_LIMIT: GPIO02 (pin 24)
Z_LIMIT: GPIO15 (pin 23)

bdring, 2019-03-22:
This might be a known issue that affects some people.

There are (2) types of dev boards. One is the NodeMCU (0.9" pin width). The other is the Espressif Dev module (1.0" pin width).

The NodeMCU has an LED on that pin that messes some people up. I use it all the time and it never causes problems, but a few people have reported the same problem you have.

[image]

There are a few solutions

1.  Switch to the other dev module
2. Solder on a stronger pullup
3. Move the x-limit to a different pin like (door, reset, start, hold) and change the mapping in the firmware.
4. Remove the LED on the dev board

bdring, 2019-03-22:
FYI: I found this discussion that explains why it might only affect a few people. The value of the resistor can vary widely.

https://esp32.com/viewtopic.php?t=5111

DavidBoman, 2019-03-22:
Fantastic! I will try one or more of the solutions as soon as I'm home! Thx,

bdring, 2019-03-22:
Great. I reached out to a friend who is an LED expert for more advice.

Slixxor, 2019-03-22:
It's recommended to use NC for limit switches rather than NO as per GRBL documentation. Also, you can use a TLP281 4ch opto isolator board instead of an R/C network. This works a treat when limit switch pins out is high 5V and limit switch grounds feeds TLP281 inputs. Then simply bridge the NC (Not connected) pin to 5V and HGND to GND. Please see example I've setup here.

The text above refers to an Uno/Nano/Mega connected GRBL controller where 5V is available. I'm still yet to test this at 3.3v hopefully the TLP281 resistance is low enough to get the LEDs to light up. fingers crossed! Worst case, you can run Vin to HVCC (generally the voltage on this should be higher, min 5V.)  

I've provided some examples below to suit an Uno the output pins are the X,Y,Z limit pins (You will need to check the diagram for the ESP32 to find matching pins):

Back of board: https://libraries.sharpsoft.com.au/sharpcnc/dev/tlp281back.jpg
Wiring diagram: https://libraries.sharpsoft.com.au/sharpcnc/dev/limitwiring.pdf

DavidBoman, 2019-03-23:
Using your suggestions @bdring I just switched to another dev-board. I have a bunch laying around so I took one that had the same pin-layout but I slightly different hardware layout and it worked great. Might have been different LED-connection or might have been a stronger pull-up - I really didn't dig into the issue as it is working now and that is the most important for me at this point. Thanks for all the help!

closed 2019-03-23 by DavidBoman