bdring/Grbl_Esp32 #109: Guru Meditation Error: Core  1 panic'ed (IllegalInstruction).

mjozwiakk, 2019-03-29:
Hi, i'm trying to run GRBL software on my ESP32, unfortunately i'm getting some issues. 

After uploading sketch, everything looks fine - Grbl starts i can see it and control from Serial Port Monitor in Arduino IDE. But after trying connect to wi-fi most of the tries ends with errors. I don't know exactly what is wrong. I'm trying to find solution in internet but with no success. Here is the communicates: 
Any ideas?  :( 

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:10312
load:0x40080400,len:6460
entry 0x400806a4

[MSG:AP Started 192.168.0.1]
[MSG:Captive Portal Started]
[MSG:HTTP Started]
[MSG:TELNET Started 23]

Grbl 1.1f ['$' for help]
Guru Meditation Error: Core 1 panic'ed (Cache disabled but cached memory region accessed)
Guru Meditation Error: Core 1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4017dd2c: bad00bad bad00bad bad00bad
Guru Meditation Error: Core 1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4017dd2c: bad00bad bad00bad bad00bad
Guru Meditation Error: Core 1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4017dd2c: bad00bad bad00bad bad00bad
Guru Meditation Error: Core 1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4017dd2c: bad00bad bad00bad bad00bad
Guru Meditation Error: Core 1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4017dd2c: bad00bad bad00bad bad00bad
Guru Meditation Error: Core 1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4017dd2c: bad00bad bad00bad bad00bad
Guru Meditation Error: Core 1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4017dd2c: bad00bad bad00bad bad00bad
Guru Meditation Error: Core 1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4017dd2c: bad00bad bad00bad bad00bad
Guru Meditation Error: Core 1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4017dd2c: bad00bad bad00bad bad00bad
Guru Meditation Error: Core 1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4017dd2c: bad00bad bad00bad bad00bad
Guru Meditation Error: Core 1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4017dd2c: bad00bad bad00bad bad00bad
Guru Meditation Error: Core 1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4017dd2c: bad00bad bad00bad bad00bad
Guru Meditation Error: Core 1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4017dd2c: bad00bad bad00bad bad00bad
Guru Meditation Error: Core 1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4017dd2c: bad00bad bad00bad bad00bad
Guru Meditation Error: Core 1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4017dd2c: bad00bad bad00bad bad00bad
Guru Meditation Error: Core 1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4017dd2c: bad00bad bad00bad bad00bad
Guru Meditation Error: Core 1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4017dd2c: bad00bad bad00bad bad00bad
Guru Meditation Error: Core 1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4017dd2c: bad00bad bad00bad bad00bad
Guru Meditation Error: Core 1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4017dd2c: bad00bad bad00bad bad00bad
Guru Meditation Error: Core 1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4017dd2c: bad00bad bad00bad bad00bad
Guru Meditation Error: Core 1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4017dd2c: bad00bad bad00bad bad00bad
Guru Meditation Error: Core 1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4017dd2c: bad00bad bad00bad bad00bad
Guru Meditation Error: Core 1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4017dd2c: bad00bad bad00bad bad00bad
Guru Meditation Error: Core 1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4017dd2c: bad00bad bad00bad bad00bad
Guru Meditation Error: Core 1 panic'ed (IllegGuru Meditation Error: Core 1 panic'ed (Unhandled debug exception)
Debug exception reason: Stack canary watchpoint triggered ()
Guru Meditation Error: Core 1 panic'ed (Guru Meditation Error: Core 1 panic'ed (Guru Meditation Error: Core 1 panic'ed (Guru Meditation Error: Core 1 panic'ed (Guru Meditation Error: Core 1 panic'ed (Unknown reason)
Guru Meditation Error: Core 1 panic'ed (Double exception)

luc-github, 2019-03-29:
What branch are you using ? Master? or WebUI ?
What version of ESP32 core do you use : 1.01 ? or latest GIT version ?

luc-github, 2019-03-30:
@mjozwiakk I am guessing per your comment here : https://github.com/luc-github/ESP3D/issues/312
this issue can be closed ? or is it still the case ?

mjozwiakk, 2019-03-30:
Yes, you can close it - the problem was probably caused by bad serial port Configuration, but i’m still mot sure about it - this „re-flash” solved also this problem :)

closed 2019-03-30 by bdring