bdring/Grbl_Esp32 #110: Information about $ codes

GandalfOp, 2019-03-29:
Hello, 
I'm trying to execute some GRBL codes from the code lines, just as if I would send them from the Universal Gcode Sender (for example:)

`if (Start_Button == 1) {
      "Send $FM"
      "Send $F"
      "Send $F=/mycode
}`

I would not do that "literally", since I'll be using interruptions on the start button, then activating a global variable from the inside, and using a watch to execute my code when this global variable triggers.

Is there any way to "Send $FM"? Otherwise, are those commands stored into memory adresses? That way I could call the adress directly. 

Thanks a lot!

GandalfOp, 2019-03-29:
I would Label this as a question, not really an issue.

bdring, 2019-03-29:
I am out of town again until Monday.  You should use the WebUI branch and see command.txt to see how SD commands work.

WebUI will merge to master soon.

Slixxor, 2019-03-30:
@GandalfOp You will need to send indiviudual commands and wait for a corresponding "ok" to be returned from the controller before sending the next command. If you need to send a sequence of commands and it isn't a G-code file job I would create flags in your software for the different groups of commands you are waiting on a response from with a global counter value to manage your current command in the sequence. It's foolproof this way.

If you regularly poll the "?" command I would recommend to add a timer to catch any command timeouts or effectively manage any commands that return responses that start with "<".

bdring, 2019-04-01:
@GandalfOp 

I think the best way to do this would be to create another channel for commands. If you look at the serialCheckTask() you will see that it can read commands from several sources (serial, bluetooth, wifi, telnet).

Adding another source like, "local_control", would allow Grbl to read commands from a queue. Buttons, etc, could add commands to the queue. 

It is important to allow Grbl to control the reading of these commands. You can't immediately push most commands, because Grbl's buffer is likely to be full or not ready to perform a new command.

You can look at the other sources as examples of how to implement a new one. It could be a part of a larger project to add a comprehensive local control panel, with LCD, etc.  I don't have time right now to do this and adding a local display goes against my basic vision of Grbl_ESP32 (The best local display is a wireless connection to your phone or PC)

GandalfOp, 2019-04-02:
Great! Thanks a lot, this is gonna be so helpful! 
Will try to add local_control so that way, if done correctly, It will automatically do what @Slixxor says. I did not think about it originally, but you're right, we need to wait until the "ok" is returned... 

Thanks a lot!

bdring, 2019-04-27:
I recently played with adding macro buttons. I wanted to simplify one of my machines for a public event. 

[image]

One button homes the machine and the other two run files off the SD card.

The code is currently in the kinematics branch. Here are the features.

- You can assign up to 4 buttons
- The buttons have a debounce feature, so you don't get accidental multiple clicks.
- Just about anything you can send via the serial port can be sent via the buttons
- Currently you need to hard code the commands, but we will probably make them adjustable, like the a macro or $N stuff

GandalfOp, 2019-04-28:
Great! Will try to run my codes that way! Thanks a lot!

closed 2019-04-28 by bdring

GandalfOp, 2019-05-09:
@bdring Hello, and sorry for answering this closed issue, but I downloaded the code, uploaded it and as you say, I could use one of the macros to run my gcode (I changed the name of the gcode on the system.cpp) But then I realized it was doing it as if it was your Polar Printer. Then tried to change the cpu map to CPU_MAP_ESP32 and define again the Macros (using unwanted GPIO and reassigning the pin == 80 on the sysyem.cpp) but it keeps printing as if it was a delta (I'm using a cartesian plotter) 

Do you have any Idea on how to change from Delta to Cartesian? Sorry again, but my programming skills are not as developed as yours...

Thanks a lot!!

GandalfOp, 2019-05-19:
Hello and sorry for answering a closed issue, but I changed the pin map to make it work on the ESP32 CNC board, and I can only make it work wiring it tho the pin 13, and it activates only while the pin is pressed (switch, not button single press) The strange thing is that when I do it on the Universal GcodeSender only with 1 press on the button it virtually activates the whole gcode, while the same program wired to the motors needs the button to be permanently pressed.

Any info I could use about that?

Thanks!