bdring/Grbl_Esp32 #115: Direction pin delay

JensHauser, 2019-04-03:
Not implemented yet, yes. I read it :-)
I already bought some DM556 step motor drivers and they require a little delay.
Is there any option to work around this issue?

bdring, 2019-04-03:
There are probably several ways to do this. 

The Grbl way is to use a timer for the delay. That uses up another precious timer, but would probably work.

The ideal ESP32 way is to use the [RMT feature](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/peripherals/rmt.html). I wrote a [blog post ](http://www.buildlog.net/blog/2017/11/esp32-sending-short-pulses-with-the-rmt/)on this a while ago. This would allow you to have an OFF_ON_OFF signal with the first 2 stages having a duration. Unfortunately I had a ton of trouble getting the RMT to play nicely in Grbl. The RMT does not like to be called from interrupts. I think this might be due to the API for the RMT. I also had trouble using the RTOS with Grbl to get around that issue. I am not an expert in the RTOS and a lot has changed in the API since I tried this, so there may be hope for the RMT.

I could probably give this another look in a couple weeks after I wrap up some other projects. If anyone knows any experts that could help with this, that would be great too.

JensHauser, 2019-04-09:
Yes, would be great.

As I have the hardware (DM556 plus 3A stepper on e.g. 1200mm X axis), I can offer to test the programming. Meanwhile I'll try learning with GRBL on Arduino, although it's not my preferred solution. ESP32 has much more advantages :-)

terjeio, 2019-04-10:
Not an expert, however this seems to be working for me:

```
void initRMT (settings_t *settings)
{
	rmt_item32_t rmtItem[2];

    rmt_config_t rmtConfig = {
        .rmt_mode = RMT_MODE_TX,
        .clk_div = 20,
        .mem_block_num = 2,
        .tx_config.loop_en = false,
		.tx_config.carrier_en = false,
		.tx_config.carrier_freq_hz = 0,
		.tx_config.carrier_duty_percent = 50,
		.tx_config.carrier_level = 0,
		.tx_config.idle_output_en = true
    };

    rmtItem[0].duration0 = settings->steppers.pulse_delay_microseconds ? 4 * settings->steppers.pulse_delay_microseconds : 1;
    rmtItem[0].duration1 = 4 * settings->steppers.pulse_microseconds;
    rmtItem[1].duration0 = 0;
    rmtItem[1].duration1 = 0;

    uint32_t channel;
    for(channel = 0; channel < N_AXIS; channel++) {

    	rmtConfig.channel = channel;

    	switch(channel) {
			case 0:
				rmtConfig.tx_config.idle_level = settings->steppers.step_invert.x;
				rmtConfig.gpio_num = X_STEP_PIN;
				break;
			case 1:
				rmtConfig.tx_config.idle_level = settings->steppers.step_invert.y;
				rmtConfig.gpio_num = Y_STEP_PIN;
				break;
			case 2:
				rmtConfig.tx_config.idle_level = settings->steppers.step_invert.z;
				rmtConfig.gpio_num = Z_STEP_PIN;
				break;
    	}
    	rmtItem[0].level0 = rmtConfig.tx_config.idle_level;
    	rmtItem[0].level1 = !rmtConfig.tx_config.idle_level;
        rmt_config(&rmtConfig);
        rmt_fill_tx_items(rmtConfig.channel, &rmtItem[0], rmtConfig.mem_block_num, 0);
    }
}
```
and 
```
// Set stepper pulse output pins
inline IRAM_ATTR static void stepperSetStepOutputs (axes_signals_t step_outbits)
{
    if(step_outbits.x) {
        RMT.conf_ch[0].conf1.mem_rd_rst = 1;
        RMT.conf_ch[0].conf1.tx_start = 1;
    }

    if(step_outbits.y) {
        RMT.conf_ch[1].conf1.mem_rd_rst = 1;
        RMT.conf_ch[1].conf1.tx_start = 1;
    }

    if(step_outbits.z) {
        RMT.conf_ch[2].conf1.mem_rd_rst = 1;
        RMT.conf_ch[2].conf1.tx_start = 1;
    }
}
```
I write directly to the RMT registers in ISR code, bypassing the API. The code can easily be changed back to the original coding style in grbl.

bdring, 2019-04-10:
@terjeio 

That looks great. I won't be able to test it until the end of the week, but it looks very similar to what I was going to try.

The API looks like it assumes you intend to change the transmitted data a lot, need to wait for the transmission to complete, and deal with multiple threads.

We want to send the exact same thing each time and the inherent timing of the steps removes all of the other issues. 

Your method looks great.

bdring, 2019-04-10:
I created a new branch for testing. I implemented the @terjeio basic code. I had a few issues getting it to compile in Grbl as he posted, so tweaked it to use global variables like Grbl tends to do.

Thanks

terjeio, 2019-04-11:
@bdring

That was quick. A minor issue in your branch: since step inversion is done at RMT init it should not be done again in the stepper ISR (line 326). Also, I forgot to include a snippet of code that assigns clock signal to the RMT peripherals - it seems it is not needed as you got it working regardless?
```
    uint32_t channel;
    for(channel = 0; channel < N_AXIS; channel++)
        rmt_set_source_clk(channel, RMT_BASECLK_APB);
```
I did not add it to initRMT since I call initRMT when settings are changed (I perform basic setup of all peripherals in a separate function).

bdring, 2019-04-11:
Thanks, I meant to remove that inversion. I have not tested any of that stuff yet. I'll add that clock stuff too.

I skipped the ganged motor features to speed up basic testing. I'll need to add that back in.

bdring, 2019-04-11:
Updated per your suggestions.

My current RMT method does not allow you to freely define and/or free step GPIO pins like before. I need to figure out a robust design method to only init needed RMT channels and refer to them in stepperSetStepOutputs()

bdring, 2019-04-24:
Changed the code to allow the same I/O and axis flexibility the master branch has.  Axes are now configured like this..

#define X_STEP_PIN      	GPIO_NUM_12
#define X_DIRECTION_PIN  	GPIO_NUM_26
#define X_RMT_CHANNEL		0

I plan to merge with master soon and have RMT be optional in config.h. This will allow it to be tested with master.

closed 2019-04-29 by bdring