bdring/Grbl_Esp32 #116: ERROR:2 using webinterinterface

Alex-CodeLab, 2019-04-05:
When trying to start a file from the SD (webinterface), I get this error:

> ERROR:2
> Missing the expected G-code word value or numeric value format is not valid.
> 

When running the same file using Telnet interface it does work as expected:

> [ESP200]
> SD card detected
> [ESP210]
> [FILE:/a.nc|SIZE:13]
> [ESP220] /a.nc
> <Idle|WPos:0.000,0.000,0.000|Bf:15,1200|FS:0,0|SD:7.69,/a.nc>

What does it mean? 
(it looks like it can't read the content of the file)

luc-github, 2019-04-05:
using command `[ESP220] /a.nc` does not process the file ? or also has error ?

Alex-CodeLab, 2019-04-05:
@luc-github  `[ESP220] /a.nc `  processes the file and no errors

luc-github, 2019-04-05:
Weird print from webUI send same command - will check that

luc-github, 2019-04-05:
Hmm, I cannot reproduce the issue
can you share the version of FW and UI ?

Alex-CodeLab, 2019-04-05:
UI: 2.1b7 / FW: 1.1f

luc-github, 2019-04-05:
can you update with latest master of GRBL_ESP32 and also webUI ? Looks you use old one , now version of  UI: 2.1b21, I suspect you use UI that use old print command `$F`

Alex-CodeLab, 2019-04-07:
I tried to install a latest GRBL_ESP32 version, but it only  gets worse:
SSD is not detected.

So, I installed the previous version again, however, now COREXY is not working, even though it is enabled in config.h ( https://gist.github.com/Alex-CodeLab/b3beae12a533779694553eb55e82805c )

luc-github, 2019-04-07:
What is SSD? SD card ?
For corexy @bdring may help

Alex-CodeLab, 2019-04-07:
yes, SD :)

btw, I just noticed ticket  #108 ; I use TMC2100 , that is not related, right?

luc-github, 2019-04-07:
I cannot say but on my side with latest FW SD is working that is why I wrote I cannot reproduce original issue.
For xycore I let bart answer

bdring, 2019-04-07:
I'll look into the CoreXY early next week.  I don't think any recent changes affected it. 

**TMC2100**. Nothing is in the code for SPI based drivers yet, but I think they can run in a legacy mode and act like A4988 and DRV8825 stepstick type drivers.

https://wiki.fysetc.com/TMC2100/ 

Are you able to run those in normal cartesian x y?

Alex-CodeLab, 2019-04-07:
@bdring yes, I had it working both corexy and cartesian.  (super silent :)

I'll investigate again next weekend

bdring, 2019-04-11:
I just did a fresh compile and upload of the master branch with CoreXY and it works normally.

Can I close this issue?

Alex-CodeLab, 2019-04-12:
turned out a broken wire was the source of my assumed corexy issue.

closed 2019-04-12 by Alex-CodeLab