bdring/Grbl_Esp32 #118: Spindle/Laser Driver?

mrmonteith, 2019-04-11:
Is there any plans to add a spindle/laser driver circuitry to the board?  The reason I ask is that it could be made into a drop-in replacement for the standard Grbl boards that are used on the desktop CNC and laser engravers that are out there.  

 I have both a CNC 3018 desktop CNC/Laser engraver and an Eleksmaker laser engraver.  As soon as I saw the ESP Grbl I was about to jump on it.  

 For myself I have no problem with basically duplicating what they use for a driver.  But I was thinking with that part of it solved on the board it could easily end up being a drop-in replacement.

 If nothing else maybe a daughter board that could serve that function.

bdring, 2019-04-11:
Most variable speed spindles and lasers just require a TTL PWM signal. Several of my controller provide that.  The signal from the ESP32 is 3.3v and that does cause problems for some larger laser power supplies. The next version of the Pen/Laser controller will have a 5V PWM.

There are no plans to directly power a laser or spindle from any of the controllers.

mrmonteith, 2019-04-11:
Have you looked at the schematic how they do it on the standard grbl controllers they have out there.  It's pretty straightforward.  I may cobble up a board when I get hold of one of the esp32 ones.  They are out currently.  Thanks.  You can close this unless you have something to add.

bdring, 2019-04-11:
I am not sure what you are talking about. Can you send a link to those boards and clarify what the "spindle/laser driver circuitry" does? Every laser power level requires a different driver. Some spindles are AC, DC, Brushless DC. What driver does all that?