bdring/Grbl_Esp32 #123: tried to install a latest GRBL_ESP32 version, but it only gets worse

alhaddar, 2019-04-16:
tried to install a latest GRBL_ESP32 version, but it only gets worse

Arduino: 1.8.8 (Windows 10), Board: "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 16MB (128Mb), 115200, None"

Build options changed, rebuilding all
sketch\web_server.cpp: In member function 'void Web_Server::end()':

web_server.cpp:220:10: error: 'class SSDPClass' has no member named 'end'

     SSDP.end();

          ^

Multiple libraries were found for "WebServer.h"
 Used: C:\Users\alhaddar\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2\libraries\WebServer
 Not used: C:\Users\alhaddar\Documents\Arduino\libraries\WebServer_tng-master
 Not used: C:\Program Files (x86)\Arduino\libraries\WebServer_tng-master
Multiple libraries were found for "DNSServer.h"
 Used: C:\Users\alhaddar\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2\libraries\DNSServer
 Not used: C:\Users\alhaddar\Documents\Arduino\libraries\DNSServer_tng-master
Multiple libraries were found for "WiFi.h"
 Used: C:\Users\alhaddar\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2\libraries\WiFi
 Not used: C:\Program Files (x86)\Arduino\libraries\WiFi
Multiple libraries were found for "SD.h"
 Used: C:\Users\alhaddar\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2\libraries\SD
 Not used: C:\Users\alhaddar\Documents\Arduino\libraries\SD
 Not used: C:\Program Files (x86)\Arduino\libraries\SD
exit status 1
'class SSDPClass' has no member named 'end'

Invalid library found in C:\Users\alhaddar\Documents\Arduino\libraries\Compressed: no headers files (.h) found in C:\Users\alhaddar\Documents\Arduino\libraries\Compressed
Invalid library found in C:\Users\alhaddar\Documents\Arduino\libraries\Compressed: no headers files (.h) found in C:\Users\alhaddar\Documents\Arduino\libraries\Compressed

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

JamesNewton, 2019-04-16:
I'll be very curious to hear what is causing this.  Current ESP32 board files are version 1.0.2 which was just released and may be causing issues. You might want to try the prior release, or search for changes to the SD library which could cause issues.

luc-github, 2019-04-16:
Update SSDP library with the one present in libraries directory, libraries are updated also

bdring, 2019-04-18:
Did this solve the problem?

It is a common issue others have had, so I started an FAQ and added this.

Can I close the issue?

closed 2019-04-19 by bdring

alhaddar, 2019-04-19:
Thank you. You can close this post.

 Upload files successfully using another computer